carla-simulator/ros-bridge #3: carla_ros_bridge error

ahmedtiko96, 2018-11-30:
I am trying to link carla with ros through this link
 https://github.com/carla-simulator/carla/tree/master/carla_ros_bridge

However I get and error when testing the installation of carla_ros_bridge when I run the test 
`rostest carla_ros_bridge ros_bridge_client.test`

this is what comes up:
**[ros_bridge_client.test] is neither a launch file in package [carla_ros_bridge] nor is [carla_ros_bridge] a launch file name**

even though I tried to source it and still fails.
Any Help??

p-buddy, 2018-12-19:
Was anyone able to resolve this? I'm seeing the same behavior, where after following all the steps (including the link being made between .../ros-bridge/src/carla_ros_bridge/ and ~/ros/catkin_ws_for_carla/src/) my system still doesn't recognize the carla-specifc ros commands and I get the error `[ros_bridge_client.test] is neither a launch file in package [carla_ros_bridge] nor is [carla_ros_bridge] a launch file name`. Could this be an issue with other catkin_environments I have installed? I've made sure that only the ~/ros/catkin_ws_for_carla/devel/setup.bash is being sourced. 

It is also not obvious to me how ros is supposed to see the necessary launch files, even after linking the folders, as in ~/ros/catkin_ws_for_carla/src/carla_ros_bridge there are no launch files, just python scripts. Am I maybe linking the wrong folder to catkin src folder?

krdindorkar19, 2019-01-17:
@p-buddy For me, whenever I got this error each time I need to run "source ~/ros/catkin_ws_for_carla/devel/setup.bash"
after that it works fine

closed 2019-04-17 by fpasch