carla-simulator/ros-bridge #13: Version mismatch of Simlater API and Client API

s7ev3n, 2018-12-20:
Hi,
I am running carla 0.9.1 binary. And I managed to successfully install ros-brigde. 
However, when I followed the tests, I met version mismatch issue.

```
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version:    = 0.9.1-55-g9a16cf9 
WARNING: Simulator API version  = 0.9.1 
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)

```

fabianoboril, 2018-12-21:
Hi,
you get this error when you run the Client using a PythonAPI version that is not compatible with the server. It seems that you use the 0.9.1 pre-compile release for the server, but for the client it seems you compile your own Carla (I assume from master branch). You should use this version also to start the server.

fabianoboril, 2018-12-21:
Duplicate of https://github.com/carla-simulator/carla/issues/1068

closed 2018-12-21 by fabianoboril

chankim, 2019-08-02:
@fabianoboril  Hi, I'm seeing the same error message. What I did was, I copied the original files to another directory and ran Update.sh (and I think this made the difference, receving newer files) and did make launch, etc. This cause that error message. Do you mean I should compile the UE4 engine(I guess no)? What should I do then? Thank you.

femelo, 2020-04-28:
I had the same problem after creating a new sensor on the server side. What solved the problem for was to rebuild after a hard-clean:
`make hard-clean && make rebuild`