carla-simulator/ros-bridge #14: Error in python API client installation

aravindSwamy94, 2018-12-20:
Hello , 
                 Im using carla 0.9.1 build version . And I need to do this  pip2 install -e .  --user --upgrade to get python API client in my workspace
But when i do that Im getting this error as shown below
    source/libcarla/libcarla.cpp:7:26: fatal error: carla/Memory.h: No such file or directory
    compilation terminated.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

Any reason why this is happening?

TixiaoShan, 2018-12-20:
Same here. No setup file in that folder.

fabianoboril, 2018-12-21:
Hi,
pip install is for users that build CARLA from source. I will update the documentation accordingly.

Regarding your issue: Currently the ROS bridge requires a CARLA version newer than 0.9.1. Either you download the current master branch from CARLA and build it from source (then pip install... is correct), or you download an intermediate build from Jenkins. In this case, you either use easy_install or simply extend your PYTHONPATH env variable to point to the egg file for your python version.

Regards,
Fabian

TixiaoShan, 2018-12-21:
@fabianoboril 
Thanks for the update! It seems that I still can't run the launch file because of an error.
```
/home/tixiao/.local/lib/python2.7/site-packages/simple_pid/PID.py:22: UserWarning: time.monotonic() not available, using time.time() as fallback. Consider using Python 3.3 or newer to get monotonic time measurements.
  warnings.warn('time.monotonic() not available, using time.time() as fallback. Consider using Python 3.3 or newer to get monotonic time measurements.')
Traceback (most recent call last):
  File "/home/tixiao/catkin_ws/src/ros-bridge-master/src/carla_ros_bridge/client.py", line 16, in <module>
    from carla_ros_bridge.bridge import CarlaRosBridge
  File "/home/tixiao/catkin_ws/src/ros-bridge-master/src/carla_ros_bridge/bridge.py", line 20, in <module>
    from carla_ros_bridge.parent import Parent
  File "/home/tixiao/catkin_ws/src/ros-bridge-master/src/carla_ros_bridge/parent.py", line 254, in <module>
    from carla_ros_bridge.vehicle import Vehicle     # pylint: disable=wrong-import-position
  File "/home/tixiao/catkin_ws/src/ros-bridge-master/src/carla_ros_bridge/vehicle.py", line 174, in <module>
    from carla_ros_bridge.ego_vehicle import EgoVehicle  # pylint: disable=wrong-import-position
  File "/home/tixiao/catkin_ws/src/ros-bridge-master/src/carla_ros_bridge/ego_vehicle.py", line 23, in <module>
    from carla import VehicleControl
ImportError: cannot import name VehicleControl
```
`python -c 'import carla;print("Success")'` gives me `Success`.

fabianoboril, 2018-12-21:
@TixiaoShan : Do you use a recent CARLA version (newer than 0.9.1)?

TixiaoShan, 2018-12-21:
@fabianoboril 
Yes. I just downloaded compiled version of 0.9.1. After using command `export PYTHONPATH=$PYTHONPATH:<path/to/carla/>/PythonAPI/<your_egg_file>`, I got a new error, the old error mentioned above is gone.
```
process[rviz-3]: started with pid [6986]
/home/tixiao/.local/lib/python2.7/site-packages/simple_pid/PID.py:22: UserWarning: time.monotonic() not available, using time.time() as fallback. Consider using Python 3.3 or newer to get monotonic time measurements.
  warnings.warn('time.monotonic() not available, using time.time() as fallback. Consider using Python 3.3 or newer to get monotonic time measurements.')
[INFO] [1545403327.348348, 0.000000]: Trying to connect to localhost:2000
[INFO] [1545403327.359960, 0.000000]: Connected
Traceback (most recent call last):
  File "/home/tixiao/catkin_ws/src/ros-bridge-master/src/carla_ros_bridge/bridge.py", line 157, in _carla_time_tick
    self.update()
  File "/home/tixiao/catkin_ws/src/ros-bridge-master/src/carla_ros_bridge/parent.py", line 156, in update
    self._create_new_children()
  File "/home/tixiao/catkin_ws/src/ros-bridge-master/src/carla_ros_bridge/parent.py", line 107, in _create_new_children
    carla_actor=actor, parent=self)
  File "/home/tixiao/catkin_ws/src/ros-bridge-master/src/carla_ros_bridge/spectator.py", line 40, in __init__
    append_role_name_topic_postfix=append_role_name_topic_postfix)  # pylint: disable=line-too-long
  File "/home/tixiao/catkin_ws/src/ros-bridge-master/src/carla_ros_bridge/actor.py", line 46, in __init__
    if carla_actor.attributes.has_key('role_name'):
AttributeError: 'Actor' object has no attribute 'attributes'
```

fabianoboril, 2018-12-21:
@TixiaoShan : You need to use a version **newer than** 0.9.1. 0.9.1 will not work (as you just saw ;-))

You can get a build from last night here:
http://158.109.9.218:8080/blue/organizations/jenkins/carla/detail/master/12/artifacts
I haven't tested it, but I guess it should work.

TixiaoShan, 2018-12-21:
@fabianoboril 
My bad, I thought 0.9.1 is the newest. I will give it a shot. Thank you so much for your time!

s7ev3n, 2018-12-24:
Hi, @fabianoboril

Thanks for your help. I have downloaded 0.9.2 pre-built, however it seems the ros_bridge still could not work with this version. It reports to use libpng16-dev which is in conflict with ros that uses libpng12-dev, it also refered in https://github.com/carla-simulator/carla/issues/901#issuecomment-449059971. 

I guess I have to build 0.9.2 without png support from source??

fabianoboril, 2018-12-24:
Hi @s7ev3n:
Try installing only the library (libpng16-16) and not the dev package. Then it should work.

Merry XMas!

TixiaoShan, 2018-12-24:
Hi @fabianoboril ,

I have downloaded the latest 0.9.2 compiled version along with the code in this repo. However, when I launch the ros bridge, I still get a similar error.
```
process[master]: started with pid [19641]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 3c1d9454-07c6-11e9-b9f0-638d049ea693
process[rosout-1]: started with pid [19654]
started core service [/rosout]
process[carla_ros_bridge-2]: started with pid [19672]
process[rviz-3]: started with pid [19679]
/home/tixiao/.local/lib/python2.7/site-packages/simple_pid/PID.py:22: UserWarning: time.monotonic() not available, using time.time() as fallback. Consider using Python 3.3 or newer to get monotonic time measurements.
  warnings.warn('time.monotonic() not available, using time.time() as fallback. Consider using Python 3.3 or newer to get monotonic time measurements.')
Traceback (most recent call last):
  File "/home/tixiao/catkin_ws/src/ros-bridge-master/src/carla_ros_bridge/client.py", line 16, in <module>
    from carla_ros_bridge.bridge import CarlaRosBridge
  File "/home/tixiao/catkin_ws/src/ros-bridge-master/src/carla_ros_bridge/bridge.py", line 20, in <module>
    from carla_ros_bridge.parent import Parent
  File "/home/tixiao/catkin_ws/src/ros-bridge-master/src/carla_ros_bridge/parent.py", line 254, in <module>
    from carla_ros_bridge.vehicle import Vehicle     # pylint: disable=wrong-import-position
  File "/home/tixiao/catkin_ws/src/ros-bridge-master/src/carla_ros_bridge/vehicle.py", line 174, in <module>
    from carla_ros_bridge.ego_vehicle import EgoVehicle  # pylint: disable=wrong-import-position
  File "/home/tixiao/catkin_ws/src/ros-bridge-master/src/carla_ros_bridge/ego_vehicle.py", line 23, in <module>
    from carla import VehicleControl
**ImportError: cannot import name VehicleControl**
[carla_ros_bridge-2] process has died [pid 19672, exit code 1, cmd /home/tixiao/catkin_ws/src/ros-bridge-master/src/carla_ros_bridge/client.py __name:=carla_ros_bridge __log:=/home/tixiao/.ros/log/3c1d9454-07c6-11e9-b9f0-638d049ea693/carla_ros_bridge-2.log].
log file: /home/tixiao/.ros/log/3c1d9454-07c6-11e9-b9f0-638d049ea693/carla_ros_bridge-2*.log
0x17562a0 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0xcbc550) ): Atte
```

fabianoboril, 2018-12-24:
@TixiaoShan : It seems you're PYTHONPATH is missing the CARLA PythonAPI. Did you consider the step to extend your PYTHONPATH according to the README?
https://github.com/carla-simulator/ros-bridge/blob/master/Readme.md

```
export PYTHONPATH=$PYTHONPATH:<path/to/carla/>/PythonAPI/<your_egg_file>
````

Having set that properly, your problem should be gone!

Merry XMas!

TixiaoShan, 2018-12-24:
@fabianoboril 
Yes, I did use that command to link it to the egg file in the PythonAPI folder.

TixiaoShan, 2018-12-25:
@fabianoboril 
It turns out that I forgot to add `PythonAPI` folder to the path. Silly mistake. Now the ros-bridge can launch with the correct PythonAPI in carla-0.9.2. However, it seems that not everything is fully implemented. Here is the window I got.

Thanks for your time! Happy Holiday!

[image]

gnykumar, 2019-02-21:
Try running the below to check if it pointing to the right version.

```
user@system:~/Carla_Sim$  python -c 'import carla; print carla'
<module 'carla' from '/home/kishor/carla/PythonClient/carla/__init__.pyc'>
```

It is pointing to the wrong Carla version, you can delete or rename the older carla version to fix the **Error**

closed 2019-04-17 by fpasch