carla-simulator/ros-bridge #15: Error in rostest ros_bridge_client.test

aravindSwamy94, 2018-12-20:
Hello , 
                Im running 0.9.1 carla build. Im able to start the server of 0.9.1 .

But when i run the carla test file , Im getting import carla failed. i.e. carla package not found error. please find the error below:

  File "/home/aravind/Desktop/Catkin_ws/src/ros-bridge-master/src/carla_ros_bridge/client.py", line 14, in <module>
    import carla
ImportError: No module named carla
[Testcase: testtestTopics] ... ok

[ROSTEST]-----------------------------------------------------------------------

[carla_ros_bridge.rosunit-testTopics/test_publish][FAILURE]---------------------
Timed out (10s) of /clock publication.
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/opt/ros/kinetic/share/rostest/nodes/publishtest", line 117, in test_publish
    self.fail('Timed out (10s) of /clock publication.')
  File "/usr/lib/python2.7/unittest/case.py", line 410, in fail
    raise self.failureException(msg)
--------------------------------------------------------------------------------

SUMMARY
 * RESULT: FAIL
 * TESTS: 1
 * ERRORS: 0
 * FAILURES: 1

Please help me on this regard. Thanks in advance

fabianoboril, 2018-12-21:
Hi,
you need to ensure that Python finds the CARLA PythonAPI.
Please see my answer in #14

fabianoboril, 2018-12-21:
#14

anshulpaigwar, 2018-12-27:
I am trying to use rosbridge with carla 0.9.2. I am having following error with rostest but unable to figure out the reason. There are lots of warning for traffic light actor and then timeout.
I tested my PythonAPI is working properly.
```
[WARN] [1545906583.770877, 0.000000]: Created Traffic-Light Actor(id=37, parent_id=0, type=traffic.traffic_light, attributes={}). Not yet fully implemented!
[WARN] [1545906583.776345, 0.000000]: Created Unsupported Traffic Actor(id=58, parent_id=0, type=traffic.speed_limit.30, attributes={})
[INFO] [1545906598.506398, 23.000000]: Shutdown requested
[INFO] [1545906598.510714, 23.000000]: Exiting Bridge
[INFO] [1545906598.633049, 23.050000]: Done
LIFETIME: WARNING! the following objects were not destructed. 
LIFETIME: SimulatorClient(localhost:2000) still alive. 
LIFETIME: AsioThreadPool still alive. 
LIFETIME: tcp client 1 still alive. 
[Testcase: testtestTopics] ... ERROR!
ERROR: max time [20.0s] allotted for test [testTopics] of type [rostest/publishtest]
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/runner.py", line 148, in fn
    self.test_parent.run_test(test)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/rostest_parent.py", line 132, in run_test
    return self.runner.run_test(test)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/launch.py", line 684, in run_test
    (test.time_limit, test.test_name, test.package, test.type))
--------------------------------------------------------------------------------

[ROSTEST]-----------------------------------------------------------------------

SUMMARY
 * RESULT: FAIL
 * TESTS: 0
 * ERRORS: 1
 * FAILURES: 0

```

anshulpaigwar, 2019-01-02:
I found the solution I did not read the documentation carefully. We have to run the python script `manual_control.py`. There was no actor spawned so no datastream so the test was showing timeout.

fabianoboril, 2019-01-03:
Ok. Thx for your PR. I just merged it. I will close this issue, as well.

closed 2019-01-03 by fabianoboril

jaypee92, 2020-02-15:
ImportError: No module named carla
[carla_ros_bridge-2] process has died [pid 8883, exit code 1, cmd /home/jdhadiellappan/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py __name:=carla_ros_bridge __log:=/home/jdhadiellappan/.ros/log/2b4d90d4-500e-11ea-89bd-e8d8d16102f5/carla_ros_bridge-2.log].
log file: /home/jdhadiellappan/.ros/log/2b4d90d4-500e-11ea-89bd-e8d8d16102f5/carla_ros_bridge-2*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete

ros bridge is not connecting in the pre build carla version,

carla 0.9.6
unreal 4.22.3
make launch is used to launch carla in CarlaUE4
 editor.
$python -c 'import carla;print("Success")' executed successfully. 

but still ros bridge is not connecting. any help?

fpasch, 2020-02-17:
@jaypee92 Please follow the instructions. E.g. extend your python path:
```export PYTHONPATH=$PYTHONPATH:<path/to/carla/>/PythonAPI/<your_egg_file>```

MichaelKlin, 2020-06-07:
@fpasch Can you please help me. I am running Carla on windows 10  and running the ros-bridge on a Ubuntu virtual machine. I set the shared folder for the VM, so PYTHONPATH will have the egg file from the shared carla folder of the host machine.
When I tried to start the launch file, I got the error. How can I fix this?
Followings are the error log:
[INFO] [1591558443.517522, 0.000000]: listening to server 10.0.2.2:2000
[INFO] [1591558443.518620, 0.000000]: using vehicle filter: vehicle.*
[INFO] [1591558443.520008, 0.000000]: Waiting for CARLA world (topic: /carla/world_info)...
[INFO] [1591558443.696462, 0.000000]: Trying to connect to 10.0.2.2:2000
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 4dc4cb81 
WARNING: Simulator API version  = 0.9.9-1-g67684c5b 
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
================================================================================REQUIRED process [carla_ros_bridge-2] has died!
process has died [pid 11694, exit code -6, cmd /home/evgen/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py __name:=carla_ros_bridge __log:=/home/evgen/.ros/log/d68c2b96-a8f5-11ea-a724-080027623d00/carla_ros_bridge-2.log].
log file: /home/evgen/.ros/log/d68c2b96-a8f5-11ea-a724-080027623d00/carla_ros_bridge-2*.log
Initiating shutdown!
================================================================================
[carla_manual_control_ego_vehicle-6] killing on exit

MichaelKlin, 2020-06-07:
@fpasch Can you please help me. I am running Carla on windows 10 and running the ros-bridge on a Ubuntu virtual machine. I set the shared folder for the VM, so PYTHONPATH will have the egg file from the shared carla folder of the host machine.
When I tried to start the launch file, I got the error. How can I fix this?
Followings are the error log:
[INFO] [1591558443.517522, 0.000000]: listening to server 10.0.2.2:2000
[INFO] [1591558443.518620, 0.000000]: using vehicle filter: vehicle.*
[INFO] [1591558443.520008, 0.000000]: Waiting for CARLA world (topic: /carla/world_info)...
[INFO] [1591558443.696462, 0.000000]: Trying to connect to 10.0.2.2:2000
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API
WARNING: Client API version = 4dc4cb81
WARNING: Simulator API version = 0.9.9-1-g67684c5b
terminate called after throwing an instance of 'std::bad_alloc'
what(): std::bad_alloc
================================================================================REQUIRED process [carla_ros_bridge-2] has died!
process has died [pid 11694, exit code -6, cmd /home/evgen/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py __name:=carla_ros_bridge __log:=/home/evgen/.ros/log/d68c2b96-a8f5-11ea-a724-080027623d00/carla_ros_bridge-2.log].
log file: /home/evgen/.ros/log/d68c2b96-a8f5-11ea-a724-080027623d00/carla_ros_bridge-2*.log
Initiating shutdown!