carla-simulator/ros-bridge #18: ros_bridge: control effect is not ideal

JackRen88, 2018-12-29:
Hello,Carla team!
Thanks for making a good simulator.When I run the control module，finding that speed controller cannot maintain constant speed .In addition,I want to know if there is a horizontal control module in the ros_bridge.

fabianoboril, 2019-01-03:
Hi,
could you provide is with some more insights, on how you run CARLA and the ros-bridge?
Thx!

JackRen88, 2019-01-10:
@fabianoboril 
I run the prebuild version of Carla,then start the ros bridge:
`roslaunch carla_ros_bridge client.launch`
and run script manual_control.py.
Test control messages:
`rostopic pub /carla/ego_vehicle/ackermann_cmd ackermann_msgs/AckermannDrive "{steering_angle: 0.0, steering_angle_velocity: 0.0, speed: 10, acceleration: 0.0,
  jerk: 0.0}" -r 10`,finding that speed controller cannot maintain constant speed.

In addition to,I also find that when starting lanuch file client.launch,appearing some errors, as follows:
`/home/rcj/.local/lib/python2.7/site-packages/simple_pid/PID.py:22: UserWarning: time.monotonic() not available, using time.time() as fallback. Consider using Python 3.3 or newer to get monotonic time measurements.
  warnings.warn('time.monotonic() not available, using time.time() as fallback. Consider using Python 3.3 or newer to get monotonic time measurements.')`  I am using Python 2.7.
How should I solved it?

fpasch, 2019-04-17:
`carla_ackermann_control` is now a separate ros node.
Please open a new issue if you're still facing issues.

closed 2019-04-17 by fpasch