carla-simulator/ros-bridge #20: Ego vehicle not moving smoothly Carla 0.9.2

anshulpaigwar, 2019-01-02:
On running the script manual_control.py along with carla_ros_bridge the ego vehicle do not run smoothly there is a tremendous amount of lag when controlled from the keyboard.

Also in my script when I spawn a vehicle in autopilot mode and set its role name as "hero", the motion of ego vehicle is not smooth. 

But when I do not set the role name of the vehicle everything works perfectly. I could not understand the problem. I tried changing the PID parameters but no improvement.

fabianoboril, 2019-01-03:
You may try if PR #22 improves the behavior.

fabianoboril, 2019-01-07:
The PR #22 is merged. I assume it fixes your problem, and we can close this issue?

closed 2019-01-07 by anshulpaigwar