carla-simulator/ros-bridge #24: Carla 0.9.2 rosbrige automatically dies when I stop my python script

anshulpaigwar, 2019-01-08:
Rosbridge dies with different errors every time I stop [ctrl + c] my python script. A similar thing happens with the scenario_runner package when a scenario finishes. Not sure if it is the desired behaviour ??

```
terminate called after throwing an instance of 'std::bad_weak_ptr'
  what():  bad_weak_ptr
[carla_ros_bridge-1] process has died [pid 12253, exit code -6, cmd /home/anshul/catkin_ws/src/carla_ros-bridge/src/carla_ros_bridge/client.py __name:=carla_ros_bridge __log:=/home/anshul/.ros/log/0a3dc748-1265-11e9-ae1d-18a905c1f8b8/carla_ros_bridge-1.log].
log file: /home/anshul/.ros/log/0a3dc748-1265-11e9-ae1d-18a905c1f8b8/carla_ros_bridge-1*.log
```
And then there are few times it doesn't dies automaically. Not sure what is the reason for this.
I also observed that while closing manual_control.py script it never dies automatically.  Is there anything I am missing / doing wrong in my script below??

```
def main():
    actor_list = []
    try:
        client = carla.Client('localhost', 2000)
        client.set_timeout(2.0)
        world = client.get_world()
        spectator = world.get_spectator()
        blueprint_library = world.get_blueprint_library()

        bp = random.choice(blueprint_library.filter('vehicle'))
        bp.set_attribute('role_name', 'hero')
        color = random.choice(bp.get_attribute('color').recommended_values)
        bp.set_attribute('color', color)
        transform = random.choice(world.get_map().get_spawn_points())
        vehicle = world.spawn_actor(bp, transform)
        actor_list.append(vehicle)
        print('created %s' % vehicle.type_id)
        vehicle.set_autopilot(True)

        lidar_bp = blueprint_library.find('sensor.lidar.ray_cast')
        lidar_bp.set_attribute('rotation_frequency', '10')
        lidar_bp.set_attribute('channels', '16')
        lidar_bp.set_attribute('range', '5000')
        lidar_bp.set_attribute('points_per_second', '222000')
        lidar_transform = carla.Transform(carla.Location(x= 0,z=2.4))
        lidar = world.spawn_actor(lidar_bp, lidar_transform, attach_to=vehicle)
        actor_list.append(lidar)
        print('created %s' % lidar.type_id)

        while True:
            time.sleep(10)

    except KeyboardInterrupt:
        print("Interrupted!")

    finally:

        print('destroying actors')
        for actor in actor_list:
            actor.destroy()
        print('done.')

if __name__ == '__main__':

    try:
        main()
    except KeyboardInterrupt:
        pass
    finally:
        print('\ndone.')
```

anshulpaigwar, 2019-01-14:
@fabianoboril could you help me find the reason behind this issue.

fabianoboril, 2019-01-14:
Well, maybe take a look into the logfiles that ros creates for each node and see if there is a good pointer to start with.

morambro, 2019-02-28:
@anshulpaigwar were you able to find the issue? I am experiencing the same error.

anshulpaigwar, 2019-03-06:
@morambro no I couldn't find the solution. RosBridge dies when I destroy the actors and it dies with different errors every time. How I am supposed to use scenario runner with ros if ros-bridge keep dying after every repetition??

fabianoboril, 2019-03-06:
The bridge is currently not designed to handle a disappearing ego car. You may add this, if you need it. Otherwise, ensure that the ego vehicle does not disappear. For the scenarios, there is a possibility to run several scenarios in sequence (see the group: option).

togaen, 2019-03-28:
@fabianoboril I would be interested in making the bridge be able to handle disappearing ego vehicles; it would greatly help my development. Can you point me to what code needs to change?

zdx3578, 2019-03-29:
@nsubiron

zlw21gxy, 2019-03-29:
It looks like it's a bug inside LibCarla, related to the  client objects used to subscribe to a sensor. I'll try to create a  minimal test reproducing this in C++. Unfortunately we won't be able to  fix it for 0.9.5.
--

@togaen @anshulpaigwar   

https://github.com/carla-simulator/carla/issues/1399

togaen, 2019-03-29:
@zlw21gxy  thanks!

fpasch, 2020-09-30:
Should be fixed with 0.9.10

closed 2020-09-30 by fpasch

kargarisaac, 2020-10-11:
> Should be fixed with 0.9.10

It's not. I have the same problem in 0.9.10.1

yasser-h-khalil, 2020-11-27:
@kargarisaac Hi, Were you able to fix it or find a walk around?

kargarisaac, 2020-11-27:
> @kargarisaac Hi, Were you able to fix it or find a walk around?

No, I didn't.