carla-simulator/ros-bridge #27: Test control messages have no effect

yangydavid, 2019-01-11:
Hi,

I'm experiencing difficulties with test control commands. I spanned a "hero" vehicle by using manual_control.py and launched the bridge ros package. I'm able to echo the topics posted by the bridge, but publishing to the /carla/ego_vehicle/ackermann_cmd topic doesn't do anything - the car only reacts to the keyboard.
I tried removing everything related to the keyboard in the manual_control.py but no luck - the ackermann_cmd still doesn't do anything.

Let me know what might be the problem.

fabianoboril, 2019-01-11:
Hi,
deactivate sending control msgs in the manual_control client, and then it works ;-)
(Line 262 in https://github.com/carla-simulator/carla/blob/master/PythonAPI/manual_control.py)

yangydavid, 2019-01-11:
@fabianoboril  Hi, I just tried what you suggested, it does disable the keyboard but the control command is still not working. Any hints on how to solve it? Thanks!

fabianoboril, 2019-01-14:
Hi,
are you publishing the ackermann msg as described in the README, i.e.
`rostopic pub /carla/ego_vehicle/ackermann_cmd ackermann_msgs/AckermannDrive "{steering_angle: 0.0, steering_angle_velocity: 0.0, speed: 10, acceleration: 0.0,
  jerk: 0.0}" -r 10`

Do you use the rosbridge with the PID controller mode, or with throttle/brake input mode? (See https://github.com/carla-simulator/ros-bridge/blob/master/config/settings.yaml --> controlmode)?

Can you provide the data published via the ego_vehicle_control_info topic? Thank you.

fabianoboril, 2019-01-14:
Hi,
and maybe try PR #31 ;-)

ruixuanz, 2020-04-12:
I can't find the solution to #31 Could you send me a screenshot of it because I encountered the same problem which is there is no control of the ego_vehicle after type in the Ackermann_control command.