carla-simulator/ros-bridge #32: Incorrect visualization of LiDAR

atinfinity, 2019-01-16:
## System information (version)
- Ubuntu 16.04 64bit
- ROS Kinetic
- CARLA v0.9.2(compiled version)
- ros-bridge https://github.com/carla-simulator/ros-bridge/commit/b054cee0d9a1d2edcc6829c81cfcce4dbca4a94a

## Detailed description
I tried to use ros-bridge and visualize LiDAR data using rviz.
But, I think that visualization of LiDAR data is incorrect.

It seems that the result of visualization is correct when the vehicle stops.

[image]

But, it seems that the result of visualization is incorrect and points of past frame remain when the vehicle moves.

[image]

I could not determine the problem which is the problem between CARLA and ros-bridge.

## Steps to reproduce
### CarlaUE4
```shell
$ cd ${CARLA_ROOT}
$ ./CarlaUE4.sh -carla-server -windowed -ResX=320 -ResY=240
```

### carla_ros_bridge
```shell
$ source ~/ros/catkin_ws_for_carla/devel/setup.bash
$ roslaunch carla_ros_bridge client_with_rviz.launch
```

### carla_ros_bridge
```shell
$ cd ${CARLA_ROOT}
$ python manual_control.py
```

And, I selected `LiDAR` as sensor.

danil-tolkachev, 2019-01-16:
You just need to set `Decay Time` to 0.

atinfinity, 2019-01-16:
@danil-tolkachev Thank you for your comment. A part of LiDAR points was drawn when I set `Decay Time` to 0. Is this correct behavior as LiDAR simulation?

[image]

danil-tolkachev, 2019-01-17:
It's another problem. Try changing fps of simulation
```
./CarlaUE4.sh -benchmark -fps=10
```

atinfinity, 2019-01-17:
@danil-tolkachev Thank you for your solution. It seems that the result of visualization is correct.

closed 2019-02-04 by atinfinity

gnykumar, 2019-02-21:
Hi,

I am trying same above steps with 0.9.2 version(pre compiled)

I am able to run it.

But while running I get below errors
For python manual_control.py
```
Traceback (most recent call last):
  File "manual_control.py", line 590, in <lambda>
    self.sensor.listen(lambda image: CameraManager._parse_image(weak_self, image))
  File "manual_control.py", line 630, in _parse_image
    self._surface = pygame.surfarray.make_surface(array.swapaxes(0, 1))
  File "/usr/lib/python2.7/dist-packages/pygame/surfarray.py", line 243, in make_surface
    return numpysf.make_surface (array)
  File "/usr/lib/python2.7/dist-packages/pygame/_numpysurfarray.py", line 368, in make_surface
    blit_array (surface, array)
  File "/usr/lib/python2.7/dist-packages/pygame/_numpysurfarray.py", line 437, in blit_array
    surface.get_buffer ().write (data, 0)
IndexError: bytes to write exceed buffer size
```

And 
For 
roslaunch carla_ros_bridge client_with_rvizaunch.launch
```
[INFO] [1550728939.884117, 49.285821]: Reconfigure Request:  speed (0.05, 0.0, 0.5),accel (0.02, 0.0, 0.05),
Traceback (most recent call last):
  File "/home/wipro16/carla_bridge_0.9.2/src/ros-bridge-0.9.2/src/carla_ros_bridge/bridge.py", line 157, in _carla_time_tick
    self.update()
  File "/home/wipro16/carla_bridge_0.9.2/src/ros-bridge-0.9.2/src/carla_ros_bridge/parent.py", line 159, in update
    actor.update()
  File "/home/wipro16/carla_bridge_0.9.2/src/ros-bridge-0.9.2/src/carla_ros_bridge/ego_vehicle.py", line 209, in update
    self.vehicle_control_cycle()
  File "/home/wipro16/carla_bridge_0.9.2/src/ros-bridge-0.9.2/src/carla_ros_bridge/ego_vehicle.py", line 445, in vehicle_control_cycle
    self.run_accel_control_loop()
  File "/home/wipro16/carla_bridge_0.9.2/src/ros-bridge-0.9.2/src/carla_ros_bridge/ego_vehicle.py", line 562, in run_accel_control_loop
    self.info.current.accel)
  File "/home/wipro16/.local/lib/python2.7/site-packages/simple_pid/PID.py", line 91, in __call__
    self._derivative = -self.Kd * d_input / dt
ZeroDivisionError: float division by zero
```

And I am not able to see any lidar information anywhere.

gnykumar, 2019-03-01:
@atinfinity 
Hi,

I have a problem, please check this for me 

https://github.com/carla-simulator/ros-bridge/issues/52#issue-415466642

gnykumar, 2019-03-05:
Hi @atinfinity @danil-tolkachev 

Thanks both of you, I resolved all those issues.
and able to bridge with compiled version of carla to get lidar data in ros

But my application is to create 3D map from lidar data in ros, here the data
I am getting is not good to generate because of it's rotation frequency and points per second

```
Blueprint attribute | Type | Default | Description
-- | -- | -- | --
channels | int | 32 | Number of lasers
range | float | 1000 | Maximum measurement distance in meters
points_per_second | int | 56000 | Points generated by all lasers per second
rotation_frequency | float | 10.0 | Lidar rotation frequency
upper_fov | float | 10.0 | Angle in degrees of the upper most laser
lower_fov | float | -30.0 | Angle in degrees of the lower most laser
sensor_tick | float | 0.0 | Seconds between sensor captures (ticks)
```
Above this information provided in carla simulator site,
Where I can modify these parameters in Compiled version and in ros bridge.

Below images are 3D map which i tried to build using ros lidar data
[image]
[image]

gnykumar, 2019-03-06:
Hi @atinfinity @danil-tolkachev 

Thanks both of you, I resolved all those issues.
and able to use bridge with compiled version of carla to get lidar data in ros

But my application is to create 3D map from lidar data in ros, here the data
I am getting is not good to generate because of it's rotation frequency and points per second

```
Blueprint attribute | Type | Default | Description
-- | -- | -- | --
channels | int | 32 | Number of lasers
range | float | 1000 | Maximum measurement distance in meters
points_per_second | int | 56000 | Points generated by all lasers per second
rotation_frequency | float | 10.0 | Lidar rotation frequency
upper_fov | float | 10.0 | Angle in degrees of the upper most laser
lower_fov | float | -30.0 | Angle in degrees of the lower most laser
sensor_tick | float | 0.0 | Seconds between sensor captures (ticks)
```

**Above this information provided in carla simulator site,
Where I can modify these parameters in Compiled version and in ros bridge.**

Below images are 3D map which i tried to build using ros lidar data
[image]
[image]

gnykumar, 2019-03-13:
Hi @atinfinity 
@danil-tolkachev 

Thanks for all

I found how to make better lidar data out

```
class CameraManager(object):
    def __init__(self, parent_actor, hud):
        self.sensor = None
        self._surface = None
        self._parent = parent_actor
        self._hud = hud
        self._recording = False
        self._camera_transforms = [
            carla.Transform(carla.Location(x=-5.5, z=2.8), carla.Rotation(pitch=-15)),
            carla.Transform(carla.Location(x=1.6, z=1.7))]
        self._transform_index = 1
        self._sensors = [
            ['sensor.camera.rgb', cc.Raw, 'Camera RGB'],
            ['sensor.camera.depth', cc.Raw, 'Camera Depth (Raw)'],
            ['sensor.camera.depth', cc.Depth, 'Camera Depth (Gray Scale)'],
            ['sensor.camera.depth', cc.LogarithmicDepth, 'Camera Depth (Logarithmic Gray Scale)'],
            ['sensor.camera.semantic_segmentation', cc.Raw, 'Camera Semantic Segmentation (Raw)'],
            ['sensor.camera.semantic_segmentation', cc.CityScapesPalette, 'Camera Semantic Segmentation (CityScapes Palette)'],
            ['sensor.lidar.ray_cast', None, 'Lidar (Ray-Cast)']]
        world = self._parent.get_world()
        bp_library = world.get_blueprint_library()
        for item in self._sensors:
            bp = bp_library.find(item[0])
            if item[0].startswith('sensor.camera'):
                bp.set_attribute('image_size_x', str(hud.dim[0]))
                bp.set_attribute('image_size_y', str(hud.dim[1]))
            elif item[0].startswith('sensor.lidar'):
                bp.set_attribute('range', '5000')
                bp.set_attribute('rotation_frequency', '40')
                bp.set_attribute('channels', '32')
                bp.set_attribute('points_per_second', '500000')
                bp.set_attribute('upper_fov', '30')
                bp.set_attribute('lower_fov', '-15')
                bp.set_attribute('sensor_tick', '0.0')
            item.append(bp)
        self._index = None

```

Added attributes of lidar as points_per_second, channels and set to better rotation frequency,
I think @atinfinity  needs to set more rotation frequency and should make Decay time as 0 in rviz.

atinfinity, 2019-03-13:
@gnykumar Thank you for your proposal! I'll try this proposal.