carla-simulator/ros-bridge #34: Running the ros-bridge cause the client delay

johnny199412, 2019-01-17:
### System information (version)

-    Ubuntu 16.04 64bit 
-    ROS Kinetic
-    CARLA v0.9.2(compiled version)

### Problem
When I open the server, by the following command

- ./CarlaUE4.sh -benchmark -fps=10  -carla-server -windowed -ResX=320 -ResY=240 

And then I use the manual_control.py to open the client

- python manual_control.py

To here I get the good performance about the simulation
But when I use the ros-bridge 

- roslaunch carla_ros_bridge client_with_rviz.launch

And then the client will be slower, when I type the "w", the vehicle will move forward a little and then be 
brake.
Sorry my english is not good enough
I want to know is any setting I forgot or can be tested or this is just a normal phenomenon

fabianoboril, 2019-01-17:
Hi,
are you using the master-branch of the ros-bridge? We made a few changes recently, to improve the usability with the manual_control client and the ros-bridge.

fpasch, 2019-04-17:
closed due to inactivity.

closed 2019-04-17 by fpasch

soulslicer, 2019-09-10:
It's still slow