carla-simulator/ros-bridge #38: undefined symbol: PyUnicode_AsUTF8String

kk2491, 2019-01-23:
Hi Guys,

I am facing the below issue while running ```roslaunch carla_ros_bridge client.launch```.

I have updated the ```PYTHONPATH``` as mentioned in the ```RlEADME``` and also I am getting success if I run ```python -c 'import carla;print("Success")'```.

```
Traceback (most recent call last):
  File "/home/user/ros/catkin_ws_for_carla/src/src/carla_ros_bridge/client.py", line 14, in <module>
    import carla
  File "/home/kishor/Carla_Sim/PythonAPI/carla-0.9.2-py3.5-linux-x86_64.egg/carla/__init__.py", line 7, in <module>
    
  File "/home/kishor/Carla_Sim/PythonAPI/carla-0.9.2-py3.5-linux-x86_64.egg/carla/libcarla.py", line 7, in <module>
  File "/home/kishor/Carla_Sim/PythonAPI/carla-0.9.2-py3.5-linux-x86_64.egg/carla/libcarla.py", line 6, in __bootstrap__
ImportError: /home/kishor/.cache/Python-Eggs/carla-0.9.2-py3.5-linux-x86_64.egg-tmp/carla/libcarla.cpython-35m-x86_64-linux-gnu.so: undefined symbol: PyUnicode_AsUTF8String
[carla_ros_bridge-2] process has died [pid 23024, exit code 1, cmd /home/kishor/ros/catkin_ws_for_carla/src/src/carla_ros_bridge/client.py __name:=carla_ros_bridge __log:=/home/kishor/.ros/log/bf8f277e-1f0b-11e9-8882-e4b97a000225/carla_ros_bridge-2.log].
log file: /home/kishor/.ros/log/bf8f277e-1f0b-11e9-8882-e4b97a000225/carla_ros_bridge-2*.log

```
Previously I had faced the similar issue, I was able to fix and posted the fix here. But now I am not able to see the post.

Could you please help me to fix the issue.

Thank you,
KK

fabianoboril, 2019-01-23:
I am not sure, but isn't ROS using Python2 by default? Try setting the PYTHONPATH to point to the egg file for Python 2.7. Does it help?

kk2491, 2019-01-23:
@fabianoboril I have tried that already.. it didn't help either... Is there anything I'm missing here!?

kk2491, 2019-01-24:
@fabianoboril I was able to fix the issue. I had to update the ```PYTHONPATH```. It was previously updated with python 3.5 egg, now I have removed 3.5 and replaced with 2.7.

Thank you

closed 2019-01-24 by fabianoboril

Derekabc, 2019-04-15:
@kk2491 I am sorry, but how to use python 3.5 egg？

kk2491, 2019-04-15:
@dosovits You just need to export the ```PYTHONPATH``` like given below:

```
export PYTHONPATH=$PYTHONPATH:/home/CARLA_0.9.5/CARLA_0.9.5/PythonAPI/carla/dist/carla-0.9.5-py3.5-linux-x86_64.egg 
```

Derekabc, 2019-04-15:
Hi, Thank you for reply. I just tried the method.  I added the PATH as following.
`PYTHONPATH=$PYTHONPATH:/home/dong/carla/PythonAPI/carla/dist/carla-0.9.5-py3.5-linux-x86_64.egg`
and then `cd ~/carla/PythonAPI/examples`, run the command `$ python automatic_control.py`
The error still remains,
`_pygame 1.9.5
Hello from the pygame community. https://www.pygame.org/contribute.html
Traceback (most recent call last):
  File "automatic_control.py", line 86, in <module>
    import carla
  File "../carla/dist/carla-0.9.5-py3.5-linux-x86_64.egg/carla/__init__.py", line 8, in <module>
  File "../carla/dist/carla-0.9.5-py3.5-linux-x86_64.egg/carla/libcarla.py", line 7, in <module>
  File "../carla/dist/carla-0.9.5-py3.5-linux-x86_64.egg/carla/libcarla.py", line 6, in __bootstrap__
  File "/home/dong/anaconda3/envs/py3.5/lib/python3.5/imp.py", line 342, in load_dynamic
    return _load(spec)
ImportError: /home/dong/.cache/Python-Eggs/carla-0.9.5-py3.5-linux-x86_64.egg-tmp/carla/libcarla.cpython-35m-x86_64-linux-gnu.so: undefined symbol: PyString_Type_
`.
Thank you, is there any wrong with it?

zwbai, 2019-06-17:
same error

Janmart10, 2020-01-20:
@Derekabc Hi, were you able to solve this issue by any chance? If so, how?

revikiran1991, 2020-01-24:
After specifying the `PYTHONPATH=$PYTHONPATH`, I am stuck with the error: `/libcarla.cpython-36m-x86_64-linux-gnu.so: undefined symbol: PyString_Type_`

fpasch, 2020-01-24:
Please verify, that your PYTHONPATH only contains exactly one carla-python-ego (either the python2 or python3 compatible one)

revikiran1991, 2020-02-10:
@fpasch  I tried using the method from [here](https://github.com/carla-simulator/carla/issues/1109#issuecomment-522797337) and added only one `$PYTHONPATH` but still I am getting the same error when I try `import carla`(`ImportError: /home/ravi/thesis/carla/Dist/CARLA_0.9.7-68-g3f6021df/PythonAPI/carla/dist/carla-0.9.7-py3.6-linux-x86_64/carla/libcarla.cpython-36m-x86_64-linux-gnu.so: undefined symbol: PyString_Type`). 
FYI: I am using a Anaconda3 virtual environment with python 3.6 (since I have no permission to install programs in root).

fpasch, 2020-02-11:
@revikiran1991 Then maybe your problem is related to Anaconda3. Have you tried venv?
In general, this issue is not related to the carla-ros-bridge. If you need further support, please ask either [here](https://github.com/carla-simulator/carla) or in discord.

revikiran1991, 2020-02-11:
@fpasch Sorry. I did not realize I was posting in carla-ros-bridge support. Thank you for your patience.