carla-simulator/ros-bridge #41: ros-bridge linter causes import error on Vehicle

jbleck, 2019-01-29:
Ubuntu 16.04, ROS Kinetic, CARLA 0.9.2.

After checking out, I ran "check.sh" to see what it turns up.

Then, when running CARLA and trying to start up the ROS bridge, per instructions on the README, I get the following error:

```
Traceback (most recent call last):
  File "/home/ubuntu/catkin_ws/src/ros-bridge/src/carla_ros_bridge/client.py", line 16, in <module>
    from carla_ros_bridge.bridge import CarlaRosBridge
  File "/home/ubuntu/catkin_ws/src/ros-bridge/src/carla_ros_bridge/bridge.py", line 20, in <module>
    from carla_ros_bridge.parent import Parent
  File "/home/ubuntu/catkin_ws/src/ros-bridge/src/carla_ros_bridge/parent.py", line 8, in <module>
    from carla_ros_bridge.vehicle import Vehicle     # pylint: disable=wrong-import-position
  File "/home/ubuntu/catkin_ws/src/ros-bridge/src/carla_ros_bridge/vehicle.py", line 8, in <module>
    from carla_ros_bridge.ego_vehicle import EgoVehicle  # pylint: disable=wrong-import-position
  File "/home/ubuntu/catkin_ws/src/ros-bridge/src/carla_ros_bridge/ego_vehicle.py", line 25, in <module>
    from carla_ros_bridge.vehicle import Vehicle
ImportError: cannot import name Vehicle
```
Git shows a bunch of files changed:
```
$ git status
	modified:   src/carla_ros_bridge/client.py
	modified:   src/carla_ros_bridge/ego_vehicle.py
	modified:   src/carla_ros_bridge/parent.py
	modified:   src/carla_ros_bridge/physics.py
	modified:   src/carla_ros_bridge/sensor.py
	modified:   src/carla_ros_bridge/vehicle.py
$ git diff src/carla_ros_bridge/ego_vehicle.py
diff --git a/src/carla_ros_bridge/ego_vehicle.py b/src/carla_ros_bridge/ego_vehicle.py
index c1bbe83..bb4f8b3 100644
--- a/src/carla_ros_bridge/ego_vehicle.py
+++ b/src/carla_ros_bridge/ego_vehicle.py
@@ -165,10 +165,9 @@ class EgoVehicle(Vehicle):
         # send control command out, if there is a ROS control publisher
         ros_control_topic = rospy.get_published_topics(namespace='/')
         if (any('/carla/ego_vehicle/ackermann_cmd' == x[0] for x in ros_control_topic) or
-            any('/carla/ego_vehicle/vehicle_control_cmd' == x[0] for x in ros_control_topic)):
+                any('/carla/ego_vehicle/vehicle_control_cmd' == x[0] for x in ros_control_topic)):
             self.carla_actor.apply_control(vehicle_control)
 
-
     def update_current_values(self):
         """
         Function to update vehicle control current values.
$ git diff src/carla_ros_bridge/vehicle.py
diff --git a/src/carla_ros_bridge/vehicle.py b/src/carla_ros_bridge/vehicle.py
index 7e2e0bb..604613f 100644
--- a/src/carla_ros_bridge/vehicle.py
+++ b/src/carla_ros_bridge/vehicle.py
@@ -5,6 +5,7 @@
 #
 # authors: Bernd Gassmann (bernd.gassmann@intel.com)
 #
+from carla_ros_bridge.ego_vehicle import EgoVehicle  # pylint: disable=wrong-import-position
 """
 Classes to handle Carla vehicles
 """
@@ -170,5 +171,5 @@ class Vehicle(Actor):
 
         self.publish_ros_message('/carla/objects', vehicle_object)
 
+
 # this import has to be at the end to resolve cyclic dependency
-from carla_ros_bridge.ego_vehicle import EgoVehicle  # pylint: disable=wrong-import-position
```

raphaellmsousa, 2019-02-04:
Man, did you find a solution? I have same problem here!

berndgassmann, 2019-02-04:
I believe you don't have the respective 'egg' file in the pythonpath (see readme).

berndgassmann, 2019-02-04:
Ah wait. Why do you have local changes?

The move of the import line is the root cause for the problem you have in running the bridge.
If I remember correctly, the check.sh only checks and doesn't change. Maybe something changed that the autopep8 is moving code around now?
But the imports have to remain where they are. At the end (that's why there is a comment) since python is not able to hide interface from implementation...

closed 2019-02-06 by berndgassmann

raphaellmsousa, 2019-02-06:
Now I have:
 from cv_bridge import CvBridge

Traceback (most recent call last):
  File "/home/raphaell/ros/catkin_ws_for_carla/src/ros-bridge/src/carla_ros_bridge/client.py", line 16, in <module>
    from carla_ros_bridge.bridge import CarlaRosBridge
  File "/home/raphaell/ros/catkin_ws_for_carla/src/ros-bridge/src/carla_ros_bridge/bridge.py", line 20, in <module>
    from carla_ros_bridge.parent import Parent
  File "/home/raphaell/ros/catkin_ws_for_carla/src/ros-bridge/src/carla_ros_bridge/parent.py", line 253, in <module>
    from carla_ros_bridge.sensor import Sensor       # noqa, pylint: disable=wrong-import-position
  File "/home/raphaell/ros/catkin_ws_for_carla/src/ros-bridge/src/carla_ros_bridge/sensor.py", line 169, in <module>
    from carla_ros_bridge.camera import Camera  # noqa, pylint: disable=wrong-import-position
  File "/home/raphaell/ros/catkin_ws_for_carla/src/ros-bridge/src/carla_ros_bridge/camera.py", line 18, in <module>
    from cv_bridge import CvBridge
ImportError: No module named cv_bridge

raphaellmsousa, 2019-02-07:
After installing cv_bridge

```
setting /run_id to f7acd508-2acc-11e9-bd9d-1831bfbbc6ee
process[rosout-1]: started with pid [18111]
started core service [/rosout]
process[carla_ros_bridge-2]: started with pid [18114]
/home/raphaell/.local/lib/python2.7/site-packages/simple_pid/PID.py:22: UserWarning: time.monotonic() not available in python < 3.3, using time.time() as fallback
  warnings.warn('time.monotonic() not available in python < 3.3, using time.time() as fallback')
Traceback (most recent call last):
  File "/home/raphaell/ros/catkin_ws_for_carla/src/ros-bridge/src/carla_ros_bridge/client.py", line 16, in <module>
    from carla_ros_bridge.bridge import CarlaRosBridge
  File "/home/raphaell/ros/catkin_ws_for_carla/src/ros-bridge/src/carla_ros_bridge/bridge.py", line 20, in <module>
    from carla_ros_bridge.parent import Parent
  File "/home/raphaell/ros/catkin_ws_for_carla/src/ros-bridge/src/carla_ros_bridge/parent.py", line 255, in <module>
    from carla_ros_bridge.vehicle import Vehicle     # noqa, pylint: disable=wrong-import-position
  File "/home/raphaell/ros/catkin_ws_for_carla/src/ros-bridge/src/carla_ros_bridge/vehicle.py", line 175, in <module>
    from carla_ros_bridge.ego_vehicle import EgoVehicle  # noqa, pylint: disable=wrong-import-position
  File "/home/raphaell/ros/catkin_ws_for_carla/src/ros-bridge/src/carla_ros_bridge/ego_vehicle.py", line 23, in <module>
    from carla import VehicleControl
ImportError: cannot import name VehicleControl

```

PR #45 (2019-02-06): Feature/fix autopep
resolves #41

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/ros-bridge/45)