carla-simulator/ros-bridge #43: Error when launching Carla with ROS

moh3th1, 2019-02-05:
**System Specifications**

- OS : Ubuntu 18.04.1 LTS (Bionic Beaver) 64bit 
- ROS: ROS Melodic
- CARLA: 0.9.3 

**Error**
I am trying to open CARLA using ROS but I faced this error when writing this command 
```
roslaunch carla_ros_bridge client.launch
``` 
[image]
`
process[carla_ros_bridge-2]: started with pid [21539]
/home/user/.local/lib/python2.7/site-packages/simple_pid/PID.py:22: UserWarning: time.monotonic() not available in python < 3.3, using time.time() as fallback
  warnings.warn('time.monotonic() not available in python < 3.3, using time.time() as fallback')
Traceback (most recent call last):
  File "/home/user/ros/catkin_ws_for_carla/src/src/client.py", line 16, in <module>
    from carla_ros_bridge.bridge import CarlaRosBridge
  File "/home/user/ros/catkin_ws_for_carla/src/src/carla_ros_bridge/bridge.py", line 20, in <module>
    from carla_ros_bridge.parent import Parent
  File "/home/user/ros/catkin_ws_for_carla/src/src/carla_ros_bridge/parent.py", line 254, in <module>
    from carla_ros_bridge.vehicle import Vehicle     # pylint: disable=wrong-import-position
  File "/home/user/ros/catkin_ws_for_carla/src/src/carla_ros_bridge/vehicle.py", line 174, in <module>
    from carla_ros_bridge.ego_vehicle import EgoVehicle  # pylint: disable=wrong-import-position
  File "/home/user/ros/catkin_ws_for_carla/src/src/carla_ros_bridge/ego_vehicle.py", line 27, in <module>
    from carla_ros_bridge.msg import CarlaVehicleControl  # pylint: disable=no-name-in-module,import-error
ImportError: No module named msg
`

raphaellmsousa, 2019-02-07:
I solved this problem using that:

```
ln -s [PATH_TO_CARLA]/carla_ros_bridge/ ~/ros/catkin_ws_for_carla/src/
source ~/ros/catkin_ws_for_carla/devel/setup.bash
rosdep update
rosdep install --from-paths ~/ros/catkin_ws_for_carla
cd ~/ros/catkin_ws_for_carla
catkin_make
```
I used the original carla ros bridge as reference since the carla simulator is missing informations like that :(

https://github.com/laurent-george/carla/tree/dev/add_ros_bridge/carla_ros_bridge

But I'm having another problems and ROS bridge still no work here, good luck

fabianoboril, 2019-02-10:
Please follow the instructions in the README.md coming with the ros-bridge. This should work. However, we did not test it with ROS Melodic. We only test with ROS Kinetic.

rrun1, 2019-02-21:
Hi, 

Something similar. I am using Carla 0.8.2 and ROS Lunar Ubuntu 16.04 LTS

When I run the 
$ roslaunch carla_ros_bridge client.launch

I get the following error.  Please advice.

SUMMARY
========

PARAMETERS
 * /carla/Episodes: 2
 * /carla/Framesperepisode: -1
 * /carla/NumberOfPedestrians: 10
 * /carla/NumberOfVehicles: 30
 * /carla/QualityLevel: Low
 * /carla/SendNonPlayerAgentsInfo: True
 * /carla/SynchronousMode: True
 * /carla/WeatherId: 1
 * /carla/host: localhost
 * /carla/port: 2000
 * /carla/sensors/camera_depth/SensorType: CAMERA
 * /carla/sensors/camera_depth/carla_settings/FOV: 90
 * /carla/sensors/camera_depth/carla_settings/ImageSizeX: 320
 * /carla/sensors/camera_depth/carla_settings/ImageSizeY: 240
 * /carla/sensors/camera_depth/carla_settings/PositionX: 1.8
 * /carla/sensors/camera_depth/carla_settings/PositionY: 0
 * /carla/sensors/camera_depth/carla_settings/PositionZ: 1.3
 * /carla/sensors/camera_depth/carla_settings/PostProcessing: Depth
 * /carla/sensors/camera_depth/carla_settings/RotationPitch: 0
 * /carla/sensors/camera_depth/carla_settings/RotationRoll: 0
 * /carla/sensors/camera_depth/carla_settings/RotationYaw: 0
 * /carla/sensors/camera_front/SensorType: CAMERA
 * /carla/sensors/camera_front/carla_settings/FOV: 90
 * /carla/sensors/camera_front/carla_settings/ImageSizeX: 320
 * /carla/sensors/camera_front/carla_settings/ImageSizeY: 240
 * /carla/sensors/camera_front/carla_settings/PositionX: 1.8
 * /carla/sensors/camera_front/carla_settings/PositionY: 0
 * /carla/sensors/camera_front/carla_settings/PositionZ: 1.3
 * /carla/sensors/camera_front/carla_settings/PostProcessing: SceneFinal
 * /carla/sensors/camera_front/carla_settings/RotationPitch: 0
 * /carla/sensors/camera_front/carla_settings/RotationRoll: 0
 * /carla/sensors/camera_front/carla_settings/RotationYaw: 0
 * /carla_autopilot: True
 * /curr_episode: 
 * /rosbag_fname: 
 * /rosdistro: lunar
 * /rosversion: 1.13.7

NODES
  /
    carla_ros_bridge (carla_ros_bridge/client.py)

 [1mROS_MASTER_URI=http://192.168.0.122:11311/ [0m
 ]2;/home/xdsim/catkin_ws/src/carla_ros_bridge/client.launch http://192.168.0.122:11311/
 [1mprocess[carla_ros_bridge-1]: started with pid [7473] [0m
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete

Wang-Dashan, 2019-03-09:
Hi,
I have also faced this problem.Have you solve it ?

auto-starting new master
process[master]: started with pid [3717]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 65386268-4210-11e9-a3c1-4ccc6a9ba76e
process[rosout-1]: started with pid [3728]
started core service [/rosout]
process[carla_ros_bridge-2]: started with pid [3734]
Traceback (most recent call last):
  File "/home/wss/Work/carla/catkin_ws_for_carla/src/src/carla_ros_bridge/client.py", line 16, in <module>
    from carla_ros_bridge.bridge import CarlaRosBridge
  File "/home/wss/Work/carla/catkin_ws_for_carla/src/src/carla_ros_bridge/bridge.py", line 18, in <module>
    from derived_object_msgs.msg import ObjectArray
ImportError: No module named derived_object_msgs.msg
[carla_ros_bridge-2] process has died [pid 3734, exit code 1, cmd /home/wss/Work/carla/catkin_ws_for_carla/src/src/carla_ros_bridge/client.py __name:=carla_ros_bridge __log:=/home/wss/.ros/log/65386268-4210-11e9-a3c1-4ccc6a9ba76e/carla_ros_bridge-2.log].
log file: /home/wss/.ros/log/65386268-4210-11e9-a3c1-4ccc6a9ba76e/carla_ros_bridge-2*.log

raphaellmsousa, 2019-03-09:
Wang-Dashan, I solved this problem doing that:

ln -s [PATH_TO_CARLA]/carla_ros_bridge/ ~/ros/catkin_ws_for_carla/src/
source ~/ros/catkin_ws_for_carla/devel/setup.bash
rosdep update
rosdep install --from-paths ~/ros/catkin_ws_for_carla
cd ~/ros/catkin_ws_for_carla
catkin_make

closed 2019-04-17 by fpasch