carla-simulator/ros-bridge #49: 'ascii' codec can't decode byte 0xc2

silgon, 2019-02-22:
As I was doing data collection with ros-bridge, I encountered the following problem with the bagfiles.

    'ascii' codec can't decode byte 0xc2

This problem is know for messages (https://answers.ros.org/question/240612/rosbag-unicodedecodeerror/).
While debugging I found out that the problem comes specifically from the message `/carla/ego_vehicle/ego_vehicle_control_info`.

Everything else seems correct. I'll check it out myself and if I find a solution I'll let you know.

silgon, 2019-02-22:
Apparently as I described in the question, it was the encoding. Here is the following command to solved the issue.
```bash
cd ros-bridge/msg
for i in *.msg;  do
  iconv -f utf-8 -t ascii//TRANSLIT -o $i $i;
done
```
I think you should add it in a commit to your repository.

mallela, 2019-02-22:
My guess is that the error is due to values like `2.34E-5` in the ROS published messages. 
My work around was changing the line in `/opt/ros/<ros-distro>/lib/python2.7/dist-packages/genpy/dynamic.py` from 
`tmp_file.file.write(full_text.encode())` to 
`tmp_file.file.write(full_text)`

silgon, 2019-02-23:
Thanks for the commend @mallela , it's nice to see your workaround. However, that would mean that `genpy/dynamic.py` is incorrect because it can cause this kind of problems. Not sure if it's really either the `*.msg` files that are incorrect (that's what I want to believe) or carla's ros-bridge implementation.

One more thing, I'm using the docker images from **osrf**, and that would mean to change that based image, or at least the genpy package.

Are you using docker or you're doing it without containers?

mallela, 2019-02-26:
Hi @silgon, No, I'm not using a docker/container.

fpasch, 2019-04-17:
@silgon Are you still facing the issue?
I executed your shell-script, but it didn't change anything.

silgon, 2019-04-17:
The script I added corrected my issue for the ros messages. But I haven't been using `ros-bridge` lately.

fpasch, 2019-04-17:
Ok. Then let's assume that the msg-files are clean now. Thanks for your reply.

closed 2019-04-17 by fpasch