carla-simulator/ros-bridge #50: Rate of publishing is low - all messages updated in vehicle.py

mallela, 2019-02-22:
Hello,

I am using ros-brige with CARLA 0.9.3 on Ubuntu 16.04

I noticed that all the messages updated by the Vehicle class have a low publish rate - even lower than depth and image publishing rates!

What is the reason this low rate?

Thanks,
Praneeta

fpasch, 2019-04-17:
Is this still happening?

mallela, 2019-04-17:
It has improved with the new ros-bridge by leaps and bounds probably due to decoupling the code for control and ego_vehicle class. But all the messages updated by the Vehicle class still have a slightly lower publishing rate that RGB/segmentation/depth (observed by using rostopic hz).

Tl;dr: Publishing rate looks reasonable. Thanks for the update!

fpasch, 2019-07-05:
With #148 this should no longer be a problem.

closed 2019-07-05 by fpasch