carla-simulator/ros-bridge #51: Multiple sensor values for the same timestamp

mallela, 2019-02-26:
Hello,

There are multiple sensor values in the bag file for the same header.stamp. I can't seem to solve this. Any ideas?

Ubuntu 16.04; Carla 0.9.3

Thanks,
Praneeta

fpasch, 2019-04-17:
which sensor are you talking about?
Is it still happening?

mallela, 2019-04-17:
I had the issue that the timestamp was exactly the same for 2 different RGB image messages. In addition in the /clock topic, the timestamps are repeating. 

Also, I wanted to ask if there was a reason/advantage that is not readily obvious to me that we use current carla time as opposed to system time to stamp the data being published? Are there any disadvantages to using system time?

slinshady89, 2019-04-30:
Probably this helps you with your issue. 
https://carla.readthedocs.io/en/latest/configuring_the_simulation/#synchronous-mode

fpasch, 2019-07-05:
With #148 carla ros bridge support synchronous mode.

closed 2019-07-05 by fpasch