carla-simulator/ros-bridge #52: Lidar data with ros

gnykumar, 2019-02-28:
https://github.com/carla-simulator/ros-bridge/issues/32#issue-399826561
#34 

Hi,
I tried steps mentioned in https://github.com/carla-simulator/ros-bridge
to run ros-bridge with compiled version of carla 0.9.3

But i am not understanding how to add lidar to get lidar point cloud data in ros
and also I am getting below error in ros client_with_rivz.launch terminal
after running manaul.py 
```
[INFO] [1551334494.393631, 260.269848]: Reconfigure Request:  speed (0.05, 0.0, 0.5),accel (0.02, 0.0, 0.05),
Traceback (most recent call last):
  File "/home/wipro16/carla_ros_ws/src/ros-bridge-0.9.3/src/carla_ros_bridge/bridge.py", line 157, in _carla_time_tick
    self.update()
  File "/home/wipro16/carla_ros_ws/src/ros-bridge-0.9.3/src/carla_ros_bridge/parent.py", line 159, in update
    actor.update()
  File "/home/wipro16/carla_ros_ws/src/ros-bridge-0.9.3/src/carla_ros_bridge/ego_vehicle.py", line 214, in update
    self.vehicle_control_cycle()
  File "/home/wipro16/carla_ros_ws/src/ros-bridge-0.9.3/src/carla_ros_bridge/ego_vehicle.py", line 450, in vehicle_control_cycle
    self.run_accel_control_loop()
  File "/home/wipro16/carla_ros_ws/src/ros-bridge-0.9.3/src/carla_ros_bridge/ego_vehicle.py", line 567, in run_accel_control_loop
    self.info.current.accel)
  File "/home/wipro16/.local/lib/python2.7/site-packages/simple_pid/PID.py", line 104, in __call__
    self._derivative = -self.Kd * d_input / dt
ZeroDivisionError: float division by zero
```

and I am able run the vehicle after running manual.py but not ros lidar data.
Can someone help Where I am missing!

System : ubuntu 16, ROS Kinetic, python2.7, pip : pip 8.1.1 from /usr/lib/python2.7/dist-packages (python 2.7)
exported : egg file from pythonAPI before running ros client launch
installed protobuf and simple pid also and libpng16-16

JustWon, 2019-02-28:
I have a same problem,,

JustWon, 2019-02-28:
I temporally solved the problem by checking #42.
the simple-pid version 0.1.5 may cause the error.
so I downgraded it to the 0.1.4.

> sudo pip install simple-pid==0.1.4

gnykumar, 2019-03-01:
> I temporally solved the problem by checking #42.
> the simple-pid version 0.1.5 may cause the error.
> so I downgraded it to the 0.1.4.
> 
> > sudo pip install simple-pid==0.1.4

@JustWon Thanks, I did same, now Erron is not coming and camera data I am able see using ros bridge, and Further am trying for lidar data, please tell me if any steps i need to follow to get lidar data

JustWon, 2019-03-01:
you can change the choice of the sensory input by ` key which is beside the the number 1 key.
or you can select the lidar input by the number 7 key.

gnykumar, 2019-03-01:
@JustWon 
Hi JustWon, 
Thanks a lot, finally i got it

Is there any way to run multiple sensors at a time?

gnykumar, 2019-03-13:
Thanks @JustWon 

I added multiple sensor at a time in manual_control.py

Now I am able to see multiple sensors at a time.

[image]

closed 2019-03-13 by gnykumar

AVA19, 2019-04-09:
@gnykumar Can you further explain how you added multiple sensors at same time in manual_control.py

gnykumar, 2019-04-16:
@AVA19 

I Can Share you modified manual_control.py.
You can try running directly or do difference checking between the original file you have and modified file 
which I am giving, you will get understand easy.

[manual_control_cam_seg_lidar.py.zip](https://github.com/carla-simulator/ros-bridge/files/3084592/manual_control_cam_seg_lidar.py.zip)

fpasch, 2019-04-16:
You might also want to have a look onto the [carla_ego_vehicle](https://github.com/carla-simulator/ros-bridge/tree/master/carla_ego_vehicle)
This now supports defining sensors via json-file.