carla-simulator/ros-bridge #60: AttributeError: 'Image' object has no attribute 'timestamp'

sietse93, 2019-03-19:
Hey people, I am using Carla0.9.4 and build it from source. When I'm using manual_control.py and launch the client.launch I get the following error: 

Traceback (most recent call last):
  File "/home/user/ros/catkin_ws_for_carla/src/ros-bridge-master/src/carla_ros_bridge/sensor.py", line 132, in _callback_sensor_data
    self.send_tf_msg()
Traceback (most recent call last):
  File "/home/user/ros/catkin_ws_for_carla/src/ros-bridge-master/src/carla_ros_bridge/child.py", line 155, in send_tf_msg
  File "/home/user/ros/catkin_ws_for_carla/src/ros-bridge-master/src/carla_ros_bridge/sensor.py", line 132, in _callback_sensor_data
        tf_msg = self.get_tf_msg()
  File "/home/user/ros/catkin_ws_for_carla/src/ros-bridge-master/src/carla_ros_bridge/camera.py", line 141, in get_tf_msg
self.send_tf_msg()
      File "/home/user/ros/catkin_ws_for_carla/src/ros-bridge-master/src/carla_ros_bridge/child.py", line 155, in send_tf_msg
    tf_msg = self.get_tf_msg()
tf_msg = super(Camera, self).get_tf_msg()
  File "/home/user/ros/catkin_ws_for_carla/src/ros-bridge-master/src/carla_ros_bridge/sensor.py", line 147, in get_tf_msg
  File "/home/user/ros/catkin_ws_for_carla/src/ros-bridge-master/src/carla_ros_bridge/sensor.py", line 147, in get_tf_msg
        tf_msg.header = self.get_msg_header()
tf_msg.header = self.get_msg_header()
  File "/home/user/ros/catkin_ws_for_carla/src/ros-bridge-master/src/carla_ros_bridge/sensor.py", line 119, in get_msg_header
  File "/home/user/ros/catkin_ws_for_carla/src/ros-bridge-master/src/carla_ros_bridge/sensor.py", line 119, in get_msg_header
        header.stamp = rospy.Time.from_sec(self.current_sensor_data.timestamp)
AttributeError: 'GnssEvent' object has no attribute 'timestamp'
header.stamp = rospy.Time.from_sec(self.current_sensor_data.timestamp)
AttributeError: 'Image' object has no attribute 'timestamp'

What am I doing wrong? I am using python 2.7.

sietse93, 2019-03-20:
ok so I removed the carla folder, downloaded the master branch and extracted the files. Ran ./Update.sh, which did download 20190319_7f836a8.tar.gz, but gave an error that it could not find some content.tar.gz file. I figured I should manually extracted the 20190319_7f836a8.tar.gz to Unreal/CarlaUE4/Content/Carla and also created a /.git directory in that directory. But when running make launch I get the following error: 

Setup.sh: llvm-6.0-ex already installed.
Setup.sh: boost-1.69.0 already installed.
Setup.sh: rpclib-v2.2.1_c1 already installed.
Setup.sh: googletest-1.8.0-ex already installed.
fatal: not a git repository (or any of the parent directories): .git
Util/BuildTools/Linux.mk:101: recipe for target 'setup' failed
make: *** [setup] Error 128

Was not sure if I should have made a new issue. Thanks in advance

fpasch, 2019-03-20:
The download issue was just fixed today. Please pull again.

sietse93, 2019-03-22:
Great, it works. Thank you!

closed 2019-03-22 by fpasch