carla-simulator/ros-bridge #63: Pedestrian control similar to vehicle control

AdriannaGmz, 2019-03-20:
Hello, I am using Carla since recently as an application where pedestrians and a ground robot can be controlled. Therefore I'd like to know if it is possible to control the pedestrian movements similarly to how the vehicles are controlled (ackerman messages). so far, I've been able to walk the pedestrians with the carla.Transform and walker.Control and I'm interested in achieving the control in similar fashion as with the ackerman messages

Thanks for the attention and best regards!

fpasch, 2019-03-21:
Do you mean controlling pedestrians via a ROS topic?

AdriannaGmz, 2019-03-21:
yes, similarly to how vehicles are controlled, ie with ackerman messages or such
I've created a node that publishes commands and the subscribed node modifies the direction vector in the "walkercontrol" , but Id be more interested in using similar controls to the ones used in the vehicles

fpasch, 2019-03-22:
I've added some initial support. Could you please give it a try:
https://github.com/carla-simulator/ros-bridge/tree/feature/pedestrian

fpasch, 2019-07-16:
Merged in https://github.com/carla-simulator/ros-bridge/commit/da8a5426973e47b81e6b5d28cc43d132b737443a

closed 2019-07-16 by fpasch