carla-simulator/ros-bridge #68: Need to publish data at a faster rate

Usaywook, 2019-03-22:
Hello,

I am working on application where my subscriber is pretty faster.
I want to publish the gnss topic at 100hz rates.
Is there a way to do that?
It seems that I need to adjust carla simulation main core. 
According to the last issue, tick interval attribute can be related to my issue.

Where could I find to adjust tick interval of gnss sensor message in carla simulation code?
and how could I adjust frequency of the topic in rosbridge code.

Although I modify  rospy.rate frequency from 200hz to 1000hz in run method function of bridge.py file, the rate of frequency of the topic has not been changed.

Thanks and regards,

fpasch, 2019-03-29:
The ROS bridge is only publishing data whenever it gets a tick from Carla.
As a hint: try using `-benchmark -fps=100` when executing CarlaUE4.sh.
And keep in mind, that the ros bridge publishes the ros-simulation-time. That might not be related to your system time (meaning, that your ros nodes should only use ros-time)

For further questions, please ask them directly within the Carla Issue tracker.

closed 2019-03-29 by fpasch

goksanisil23, 2021-01-22:
Hi @fpasch ,
I understand that the ros-bridge rate is driven by the carla world tick. But as @Usaywook mentioned, there are different sensors on the vehicle which requires different update rates, most of them which exceeds the frame rate in Carla. (100fps frame rate in Carla is not possible via 2080Ti that I'm using, for instance)
I believe there is substepping happening for physics, which can produce data in a higher rate in Unreal corresponding to positional sensors data, but I'm not sure how to go about it.
Do you know if Carla team has a plan to support sensor data that has higher update rate than the frame rate?

johnren-code, 2024-11-07:
Hello, have you solved it?