carla-simulator/ros-bridge #90: Add dynamic capabilities to vehicle_info message

togaen, 2019-04-05:
It doesn't look like information about the ego vehicles dynamic capabilities are transmitted to ROS. Would it possible to add them to the vehicle info message? Things like min/max bounds on speed/acceleration/jerk. I know that exact bounds can be tricky due to state dependency, but even just having approximate global bounds would be useful. What I'd like is primarily a sanity check: if my controller wants to command the vehicle to accelerate with 999m/s^2, I'd like to know that that's not physically possible.

fpasch, 2019-04-05:
This completely depends on the capabilities of the Carla Python API. Could you please check if the additional info is available there? If not, please file an issue there.

You might also be interested in `carla_ackermann_control` which calculates some limits:
https://github.com/carla-simulator/ros-bridge/blob/master/carla_ackermann_control/src/carla_ackermann_control/carla_control_physics.py

togaen, 2019-04-05:
Thanks, I'll check on that.

togaen, 2019-04-05:
@fpasch The limits computed in the `carla_ackermann_control` package you linked are hard-coded. Do you know if they guaranteed somehow to be the same values that the simulator uses?

fpasch, 2019-04-08:
The values were hardcoded as we couldn't access them via Python API at the time of implementation.
We tried to keep the calculation as transparent as possible to be able to make use of Carla-provided values in the future.
At the moment, they're not guaranteed to match with Carla.

togaen, 2019-04-08:
@fpasch Thanks, I opened an issue to get them: https://github.com/carla-simulator/carla/issues/1500