carla-simulator/ros-bridge #92: Import Error with agents package [carla_waypoint_publisher]

SIlvaMFPedro, 2019-04-05:
Good evening. I installed the "agents" package with the command:
 - sudo pip install agents
But for some reason my Pycharm says that this package does not have a navigation folder.
[image]
Did i install the right package? Or do i have to installed the package some other way?

Thank you

Pedro Silva - 72645

SIlvaMFPedro, 2019-04-05:
Nevermind. I figured it out.

closed 2019-04-05 by SIlvaMFPedro

vehxianfish, 2020-11-25:
Hi,I also meet this question,not find any solution.Can you tell me how to solve this question?thank you.

jishenghuang, 2024-09-12:
I encountered the same problem. How did you solve it?