carla-simulator/ros-bridge #96: ROS bridge to Baidu Apollo

volprjir, 2019-04-10:
Hi everyone. I would like to connect Carla with Baidu Apollo project. There used to be some kind of bridge for older version of Carla from nemodrive (https://nemodrive-docs.readthedocs.io/projects/nemodrive-apollo/en/latest/index.html) but even this one didn't work perfectly. Is here anyone who already successfuly implemented connection between Carla 0.9.x to the Apollo? Can anyone help me how to do it, please? Thank you

fpasch, 2019-04-11:
That should be doable. It all comes down to replacing the publishers/subscribers by the ones that are required for Apollo Cyber RT framework.
I could give you some hints. Please join our [Discord](https://discord.gg/8kqACuC) and post your question within the ROS channel.

fpasch, 2019-04-15:
[image]
The connection between the containers is already supported by Carla.

fpasch, 2019-04-17:
There is now a development branch that separates the binding.
https://github.com/carla-simulator/ros-bridge/tree/feature/apollo

As an example you can use the LogBinding within Apollo context to see what kind of data is received from Carla.

mickeyouyou, 2019-08-28:
thank you , is this the api to carla?  maybe we can have a try

fpasch, 2020-09-30:
https://github.com/AuroAi/carla_apollo_bridge provides a apollo bridge.

closed 2020-09-30 by fpasch

dhdiagram4011, 2022-04-04:
hello @fpasch  

When running bridge with python2 version, the following message occurs, and when running bridge with python3 version, the following message occurs.
[오후 4:42]
root@eb4dae4b00c6:~/carla_apollo_bridge/examples# python run_bridge.py 
Traceback (most recent call last):
  File "run_bridge.py", line 10, in <module>
    from carla_cyber_bridge.bridge import CarlaCyberBridge
ImportError: No module named carla_cyber_bridge.bridge
[오후 4:42]
root@eb4dae4b00c6:~/carla_apollo_bridge/examples# python3 run_bridge.py 
Traceback (most recent call last):
  File "run_bridge.py", line 7, in <module>
    from cyber_py import cyber
  File "/apollo/cyber/python/cyber_py/cyber.py", line 44, in <module>
    _CYBER_INIT = importlib.import_module('_cyber_init')
  File "/usr/lib/python3.6/importlib/init.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: dynamic module does not define module export function (PyInit__cyber_init)

anaivebird, 2022-06-26:
> hello @fpasch
> 
> When running bridge with python2 version, the following message occurs, and when running bridge with python3 version, the following message occurs. [오후 4:42] root@eb4dae4b00c6:~/carla_apollo_bridge/examples# python run_bridge.py Traceback (most recent call last): File "run_bridge.py", line 10, in from carla_cyber_bridge.bridge import CarlaCyberBridge ImportError: No module named carla_cyber_bridge.bridge [오후 4:42] root@eb4dae4b00c6:~/carla_apollo_bridge/examples# python3 run_bridge.py Traceback (most recent call last): File "run_bridge.py", line 7, in from cyber_py import cyber File "/apollo/cyber/python/cyber_py/cyber.py", line 44, in _CYBER_INIT = importlib.import_module('_cyber_init') File "/usr/lib/python3.6/importlib/init.py", line 126, in import_module return _bootstrap._gcd_import(name[level:], package, level) ImportError: dynamic module does not define module export function (PyInit__cyber_init)
> 
> Thank you very much~

Same error

xqgex, 2023-03-02:
The Docker base image of `auroai/cyber:v5.0.0` is this file:
https://github.com/ApolloAuto/apollo/blob/v5.0.0/docker/build/cyber.x86_64.dockerfile
Apollo version 5.0.0!!! and then two commands:
```sh
git clone --single-branch -b v5.0.0 https://github.com/apolloauto/apollo /apollo
cd /apollo && ./apollo.sh build_cyber
```
I assume that you won't get too far with `auroai/cyber:v5.0.0`