carla-simulator/ros-bridge #100: Get sim frequency/time of next simulator tick

togaen, 2019-04-11:
Is information about the simulation frequency published through the ROS bridge? My controller needs to know when the next simulation cycle is happening.

fpasch, 2019-04-12:
Are you sure that you need that?
By using the fixed time step mode in Carla (e.g. `-benchmark -fps=20`) you ensure, that you get 20 ticks per second. That will lead to incrementing the ROS time by steps of 0.05s.
You just have to ensure, that your nodes only use the ROS time.

togaen, 2019-04-13:
@fpasch Yes, my nodes need to know how long each time step is. I can tell them manually, I just thought that information might be available programmatically somehow. (And I'd rather not reverse engineer the information from the `/clock` topic.) It's not a big deal, it would just be nice to have.

fpasch, 2019-08-02:
In synchronous mode, the delta-time is now available in [CarlaStatus.msg](https://github.com/carla-simulator/ros-bridge/blob/master/carla_msgs/msg/CarlaStatus.msg)

closed 2019-08-02 by fpasch

togaen, 2020-04-24:
@fpasch I'm a little late, but thanks for this!