carla-simulator/ros-bridge #122: visualize two lidar sensors' data in rviz at the same time [Carla 0.9.5]

bigsheep2018, 2019-05-28:
As the title mentioned, I checked several issue like #52 #51 and cannot find an exact way for visualizing multiple lidar sensors' data in rviz at the same time.

Please give a solution for this problem.
Thanks in advance.

bigsheep2018, 2019-05-28:
Update.
Update.
I am currently able to spawn multiple lidar in manual_control.py and get lidar data from two lidar sensor through ros_bridge. But it seems the data from two lidar are sending to the same topic which leads to some conflict in rviz visualization.

One of the lidar's transformation is not correct and the points is flashing in rviz.
May I know if it is possible to set a different topic for different sensor instance? or is there any other way that work this problem out

[image]

fpasch, 2019-06-06:
Are you using [carla_ego_vehicle](https://github.com/carla-simulator/ros-bridge/tree/master/carla_ego_vehicle) to set up vehicle and sensors? You need to set different sensor names., then it should work as expected.

slyer0009, 2019-06-26:
To get different Topic names I use this little workaround with a global variable counting up each time a new sensor is created.
lidar.py in src/carla_ros_bridge
`class Lidar(Sensor):

    """
    Actor implementation details for lidars
    """
    counter = 0

    def __init__(self, carla_actor, parent, topic_prefix=None, append_role_name_topic_postfix=False):
        """
        Constructor

        :param carla_actor: carla actor object
        :type carla_actor: carla.Actor
        :param parent: the parent of this
        :type parent: carla_ros_bridge.Parent
        :param topic_prefix: the topic prefix to be used for this actor
        :type topic_prefix: string
        :param append_role_name_topic_postfix: if this flag is set True,
            the role_name of the actor is used as topic postfix
        :type append_role_name_topic_postfix: boolean
        """
        Lidar.counter += 1
        if topic_prefix is None:
            topic_prefix = 'lidar'+str(Lidar.counter)
        super(Lidar, self).__init__(carla_actor=carla_actor,
                                    parent=parent,
                                    topic_prefix=topic_prefix,
                                    append_role_name_topic_postfix=append_role_name_topic_postfix)`

Hope it works for you too!

closed 2019-06-28 by fpasch