carla-simulator/ros-bridge #147: Unable to get messages from /carla/ego_vehicle/camera/rgb/front/image_color

strider175, 2019-07-05:
Hello everyone,

I am using CARLA 0.9.5., ROS Kinetic, Ubuntu 16.04, Carla Ros Bridge 0.9.5. for my graduate project in which I am supposed to to control my vehicle by using ROS.
My problem is that when i use ros tutorial code for subscribing to rostopic /carla/ego_vehicle/camera/rgb/front/image_color and when I list rostopic info i can see that my node is subscribed to rostopic but my callback function is never called. So i used rostopic hz to check whether messages are sent and i get following output where it seems that subscriber is not sending messages to my node:
[image]

I am really stuck on this step for a while and I hope that someone can help me.
Thank You in advance.

fpasch, 2019-07-05:
I'm not 100% sure, but `rostopic hz` might not be working with simulated time (as reported by your rostopic call).

Please first ensure, that you get any data at all. E.g. by `rostopic echo /clock`.
Then check with rviz.

strider175, 2019-07-05:
When I type `rostopic echo /clock` I get following:
[image]
I tried setting use_sim_time to false but then nothing happens.
Also in rviz there are no images received
[image]

fpasch, 2019-07-09:
Do you still face the issue?

strider175, 2019-07-10:
Yes, I am still unable to resolve the issue.

GlinZhu, 2019-07-13:
I had the exact same issue with you, but I was working on Udacity self-driving car project, haven't worked on carla simulator before. and you are right, the reason is that the subscriber is not sending the images to your node. I fixed the issue by checking the camera toggle button in the simulator, which may not be quite helpful for you, but it may give you some insights..

fpasch, 2019-07-15:
If you run 
```
roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch
 ```

Do you see an image in the pygame window? If yes, that's received via the camera topic.

strider175, 2019-07-17:
When I run `roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch` I get this window without image:
[image]

fpasch, 2019-07-17:
So you issue is not related to camera-data at all.
Please precisely follow the documentation on installation and execution.

Please also ensure, that you're using Carla 0.9.6 with current master (or carla 0.9.5 with tagged version 0.9.5.1)
If the error remains have a look at the output during execution.

I'll close that issue for now.

closed 2019-07-17 by fpasch