carla-simulator/ros-bridge #152: Support IMU

atinfinity, 2019-07-08:
Does `ros-bridge` have plan to support IMU? Because, some ROS package need IMU data.
The `carla-simulator` has the following API according to the [document](https://github.com/carla-simulator/carla/blob/master/Docs/python_api_tutorial.md). So, I think that `ros-bridge` can handle IMU.

- get_acceleration
- get_angular_velocity
- get_transform

fpasch, 2019-07-09:
It's not planned, but you should be able to easily implement it yourself:

Have a look at [ego_vehicle.py](https://github.com/carla-simulator/ros-bridge/blob/master/carla_ros_bridge/src/carla_ros_bridge/ego_vehicle.py#L96).There you can see how the messages are created.

Feel free to send a pull request.

atinfinity, 2019-07-09:
@fpasch Thank you for your comment. I'll try it.

atinfinity, 2019-07-15:
@fpasch I tried to implement IMU publisher in [vehicle.py](https://github.com/carla-simulator/ros-bridge/blob/0.9.5.1/carla_ros_bridge/src/carla_ros_bridge/vehicle.py).

I think that [IMU message](http://docs.ros.org/melodic/api/sensor_msgs/html/msg/Imu.html) needs the following parameters.  

* orientation
* angular_velocity
* linear_acceleration

So, I have a question about Python API.
Does the following API return a value in the `map` coordinate system?

* carla.Actor.get_transform
* [carla.Actor.get_acceleration](https://carla.readthedocs.io/en/latest/python_api/#carla.Actor.get_acceleration)
* [carla.Actor.get_angular_velocity](https://carla.readthedocs.io/en/latest/python_api/#carla.Actor.get_angular_velocity)

fpasch, 2019-07-15:
Yes. They're in world coordinates (i.e. in map frame).

Please have a look [here](https://github.com/carla-simulator/ros-bridge/blob/64dada65c91a2ad2e4e2500180374cfc04731f66/carla_ros_bridge/src/carla_ros_bridge/ego_vehicle.py#L103) on how to transform the values into local frame.

atinfinity, 2019-07-15:
@fpasch Thank you for your advice. And, do `orientation` and `angular_velocity` need similar transformation?

fpasch, 2019-07-16:
Please have a look [here](https://github.com/carla-simulator/ros-bridge/blob/64dada65c91a2ad2e4e2500180374cfc04731f66/carla_ros_bridge/src/carla_ros_bridge/transforms.py#L210) and [here](https://github.com/carla-simulator/ros-bridge/blob/64dada65c91a2ad2e4e2500180374cfc04731f66/carla_ros_bridge/src/carla_ros_bridge/transforms.py#L310)

Finally, I highly recommend to verify the outcome within e.g. rviz.

atinfinity, 2019-07-16:
@fpasch Thank you for your advice.

## linear_acceleration
I implemented and visualized linear_acceleration in rviz. Both cases seem to be valid.

### move forward
[image]

### sudden braking
[image]

## orientation
I implemented the following code to calculate orientation of IMU.

```python
rotation = self.carla_actor.get_transform().rotation
imu.orientation = transforms.carla_rotation_to_ros_quaternion(rotation)
```

And, I tried to check value using `rostopic echo`. Both cases are running on the flat ground.
But, I think that the result of Case2 is incorrect. Could you give me your comment?

### Case#1
```
orientation: 
  x: -1.18190891456e-06
  y: -6.08120009148e-06
  z: 0.00128763802398
  w: 0.999999170975
```

### Case2
```
orientation: 
  x: 0.000299430007648
  y: -0.000205472390706
  z: 0.999886294096
  w: 0.0150753773212
```

fpasch, 2019-07-16:
Nice to see progress.

The orientation you get from `get_transform()` is in world frame. You have to transform it.
Please have a look at `carla_vector_to_ros_vector_rotated()`.

atinfinity, 2019-07-17:
@fpasch Thank you for your advice. How do I need to apply the transformation to get the orientation of vehicle?

fpasch, 2020-01-14:
implemented on carla-side and #215

closed 2020-01-14 by fpasch