carla-simulator/ros-bridge #162: missing rostopics after ros_bridge

kk2491, 2019-07-19:
Hi Guys,

I have followed the steps for the ```rosbridge``` as given in the README. I was able to get the rosbridge up too. However when I do ```rostopic list``` I am able to see only the below topics.

```
/carla/status
/clock
/rosout
/rosout_agg
/tf
```

Is there any steps I am doing wrong? Could you please let me know how could I get the control related commands. I want to send the values manually to control commands to check the car behavior.

Thank you,
KK

fpasch, 2019-07-19:
With latest master of ros-bridge, please use Carla 0.9.6.
If you want to have ros-bridge running with Carla 0.9.5, please use the tagged ros-bridge version 0.9.5.1.

kk2491, 2019-07-20:
Yep.. Thank you.. that was the problem... Now I am using ros-bridge version 0.9.5.1 and it is working fine.

closed 2019-07-22 by fpasch