carla-simulator/ros-bridge #173: Add receiver in bridge to enable debug features

cmpute, 2019-08-09:
Could you add some feature to receive ROS messages and display them in Carla? Specifically display visualization_msgs/Marker and visualization_msgs/MarkerArray in Carla spectator. [Several marker primitives](http://wiki.ros.org/rviz/DisplayTypes/Marker) are already in the debug helper function: draw_line, draw_arrow, draw_text, draw_box.

Additionally, is there any approach to display text from draw_text in rviz? I noticed that the documentation said that the text is only visible in server side, however I found a class called spectator in the ros_bridge package.

fpasch, 2019-08-13:
Could you please give pull request #176 a try and comment if it's working for you?

cmpute, 2019-08-13:
Cool! I tested it and it works now! Thanks a lot!

cmpute, 2019-08-13:
I found a small bug. It seems that the y coordinate is not converted back to CARLA system (no negative sign is added for y values). Thus the result boxes and arrows shown in CARLA is opposite

cmpute, 2019-08-14:
Another problem is that does Carla support delete existing debug elements? I use similar approach as yours to create markers, however it seems that Marker.MODIFY and Marker.DELETE are not actually supported.

fpasch, 2019-08-14:
You're warmly welcome to contribute. Would you be able to send a fix for the Carla coordinate conversion?

Regarding the removal of debug markers:
It seems not to be supported by CARLA itself. Please file an issue there addressing this feature.

closed 2019-08-15 by cmpute