carla-simulator/ros-bridge #185: Sync Mode: Understanding it

soulslicer, 2019-08-25:
I am trying to figure out Sync Mode.

I assume its a gurantee that every frame will be processed. However, when I set sync mode on, and i set the fixed_delta_seconds flag to 0.02 (so that my camera runs at 50fps), the entire scene/time slows down.

However, disabling sync mode allows it to run at normal speed, and I still get 50fps for my camera stream.

Whats happening here

fpasch, 2019-08-27:
In sync-mode the ros-bridge is controlling the tick(). It waits for every sensor to create data and then proceeds with the next tick.

In default-mode the ros-bridge is passive. It updates the available actors whenever It gets called by the tick()-callback. And it publishes sensor data whenever the sensor-listen()-callback is called. These two callback can be executed at different points in time. For a camera the callback is delayed because the sensor-data creation is taking time. That means you might see data for a previous frame  published.

Therefore synchronous mode is slower. But you get exactly the sensor-data for that frame.

fpasch, 2020-03-09:
Setting ```synchronous_mode_wait_for_vehicle_control_command``` to ```false``` should reduce the impact.
Because of the required synchronisation, there might still be some frame rate reduction.

closed 2020-03-09 by fpasch