carla-simulator/ros-bridge #189: Arch support

yeduzhouzi, 2019-09-02:
I installed ros melodic on Arch since it's now officially supported, but I cannot find a way to properly install ros-bridge with it. Is there a way to do so or it's not going to be supported?

cmpute, 2019-10-12:
You can build this package in the same ways as other packages. For example just `catkin_make` and `catkin_make` install should work.

yeduzhouzi, 2019-10-12:
Thanks for your reply @cmpute , but the real problem is not about the `catkin_make` , it is that when you run `rosdep install --from-paths src --ignore-src -r`, you will get error message that `apt-get` should not be used, since the package manager of arch linux is pacman.

cmpute, 2019-10-12:
> Thanks for your reply @cmpute , but the real problem is not about the `catkin_make` , it is that when you run `rosdep install --from-paths src --ignore-src -r`, you will get error message that `apt-get` should not be used, since the package manager of arch linux is pacman.

So actually you can just install the dependent packages by yourself (using `pacman`) and then directly use `catkin_make` to build.

yeduzhouzi, 2019-10-12:
Thanks again for your reply @cmpute ,  but do you mean that I can manually use `pacman` to locally install those packages in the `ros_bridge` directory that are supposed to be installed with the `rosdep install --from-paths src --ignore-src -r` command which actually is using `apt-get` in default? Sorry but I'm not quite familiar with Linux yet.

cmpute, 2019-10-12:
> Thanks again for your reply @cmpute , but do you mean that I can manually use `pacman` to locally install those packages in the `ros_bridge` directory that are supposed to be installed with the [...]

Yes, that's what I mean. And specifically, if you run `rosdep check --from-paths src`, then you can know which package you need to install with pacman

yeduzhouzi, 2019-10-12:
@cmpute okay I got it, thanks very much for the help.

closed 2019-10-12 by yeduzhouzi