carla-simulator/ros-bridge #190: Cannot CMake ros-bridge

yanym, 2019-09-04:
Hi there,

I notice that the latest ros-bridge only supports the Carla which version is newer than 0.96. However, I am still in 0.95 so I checkout the release version of ros-bridge (0.9.5.1). It is quite confusing that there is an error happened. I have tried to figure it out for a long time but I didn't make it. Thank you so much.

```
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by
  "derived_object_msgs" with any of the following names:

    derived_object_msgsConfig.cmake
    derived_object_msgs-config.cmake

  Add the installation prefix of "derived_object_msgs" to CMAKE_PREFIX_PATH
  or set "derived_object_msgs_DIR" to a directory containing one of the above
  files.  If "derived_object_msgs" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  ros-bridge/carla_ros_bridge/CMakeLists.txt:5 (find_package)

-- Could not find the required component 'derived_object_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "derived_object_msgs" with any of the following names:

    derived_object_msgsConfig.cmake
    derived_object_msgs-config.cmake

  Add the installation prefix of "derived_object_msgs" to CMAKE_PREFIX_PATH
  or set "derived_object_msgs_DIR" to a directory containing one of the above
  files.  If "derived_object_msgs" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  ros-bridge/carla_ros_bridge/CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/xu/carla-ros-bridge/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/xu/carla-ros-bridge/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

soulslicer, 2019-09-10:
just compile this

http://wiki.ros.org/derived_object_msgs

fpasch, 2019-09-18:
Please follow the documentation:
```
#install required ros-dependencies
rosdep update
rosdep install --from-paths src --ignore-src -r
```

closed 2019-09-18 by fpasch