carla-simulator/ros-bridge #192: Ros Bridge causes client delay

soulslicer, 2019-09-10:
This problem still persists!

https://github.com/carla-simulator/ros-bridge/issues/34

Launch Carla, then launch `python manual_control.py`. Runs smooth.

Then launch the ros_bridge, and everything grinds to a halt. Close the ros-bridge and it speeds up again.

This makes the ros_bridge unusable!

soulslicer, 2019-09-10:
It seems the very action of making a camera actor listen to data

`self.actor.listen(self._callback_sensor_data)`

Causes everything to grind to a halt. The more sensors we add, the slower it gets.

I think the problem is that using Python is slow

soulslicer, 2019-09-10:
Even implementing a toy script without carla, just the very act of subscribing to a camera causes this.

soulslicer, 2019-09-10:
Turns out th

closed 2019-09-10 by soulslicer

thillRobot, 2021-11-02:
I am seeing the same behavior in CARLA 9.12 in python 3.6. Is there a know fix to this?

thillRobot, 2021-11-08:
OK @soulslicer , after doing some reading it appears to be a Python2 Python3 compatibility issue. I am attempting to fix this by compiling `tf2` and `cv_bridge` with Python 3.7 with `venv`. It has taken some fiddling, but I think I am close. Here are my [notes on github](https://github.com/thillRobot/carla_simulator/blob/carla-py37/docs/ros_bridge.md) .