carla-simulator/ros-bridge #193: Waypoints to vehicle commands

mxpanoff, 2019-09-10:
Hello, 
I have recently installed the latest versions of Carla and Carla ROS-bridge and encountered a small issue: although both manual override and autopilot controls result in correct movement, I cannot make the ego-vehicle move by setting 2d nav goals thorugh rviz. I launch the ros-bridge, ego-vehicle, rviz, and waypoint publisher from the launch files in this repo, and am able to successfully create an ego-vehicle in carla through the ros bridge, as well as receive sensor input and control it, but I cannot make it move through rviz. I have tried different combinations of autopilot, manual and ackermann control, but nothing works. After examining the rqt_graph I see that nothing is publishing to the /carla/ego-vehicle/vehicle_control_cmd topic, although the goal and waypoint topics are active, and respond to changes in rviz. My guess is that I am forgetting to launch a .launch file of some sort that acts as a controller, turning waypoints into commands. Any help/suggestions would be appreciated

fpasch, 2019-09-18:
Please execute:
```
roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch
```
And then open rviz and set the target.

mxpanoff, 2019-09-18:
I'm getting 

> Timeout (1s) while waiting for vehicle control commands. Missing command from actor ids [92]

repeatedly printing on the terminal where I launched the ros_bridge, and the vehicle still doesn't move. I can confirm that I've set a 2d nav goal and a pose estimate, and have tried this both with and without the waypoint publisher launched as well.

fpasch, 2019-09-18:
Please disable `synchronous_mode` to get rid of the warning.

mxpanoff, 2019-09-18:
The vehicle doesn't move with synchronous mode set to false, that only seems to hide the warning

fpasch, 2019-10-02:
After re-reading your initial post:
You're simply missing the node(s) that control the vehicle. Normally, you would want to create your own implementation for that task. 

Optionally you could make use of an existing AD stack. Please have a look at [carla-autoware](https://github.com/carla-simulator/carla-autoware).

closed 2019-10-02 by fpasch