carla-simulator/ros-bridge #194: Cannot convert depth image messages to OpenCv Mat

javieryu, 2019-09-14:
I am trying to use a depth image, but when I pass the images from the ros-bridge depth_image topic to the CvBridge in the following way:

`cv_dep = self.cv_bridge.imgmsg_to_cv2(depth_image, "32FC1")`

I get a "std::bad_cast" error.

Rviz seems to have no problem converting the images and displaying them.

fpasch, 2019-09-18:
The conversion might be incorrect. Would you be able to verify it?

It's located here:
https://github.com/carla-simulator/ros-bridge/blob/master/carla_ros_bridge/src/carla_ros_bridge/camera.py#L245

soulslicer, 2019-09-18:
Dont use 32fc1 use passthrough

javieryu, 2019-09-18:
@fpasch I copied the conversion directly from the camera.py.

@soulslicer I have also tried: `cv_dep = self.cv_bridge.imgmsg_to_cv2(depth_image, desired_encoding="passthrough")`. That also failed and threw a `std::bad_cast`.

chipkajb, 2019-10-29:
I am running into this same problem. But even going back to an older version isn't helping. @javieryu What version did you use to get it to work? If the most recent version causes this error, perhaps this issue should be re-opened?

javieryu, 2019-10-29:
I am on commit 294258069310575a249835a37f8cae20a3019b68. I got pretty much no where trying to debug this one. I think it was a problem with my ROS catkin workspace, but cannot confirm that. I fixed it by wiping everything clean, and reinstalling. Creating a new catkin-ws and then reinstall all of the components I was using for my project. 

I only ever experienced this error when upgrading my carla-ros-bridge. Is this the same for you or did you catch this out of the box? If it's not then we should reopen the topic.

chipkajb, 2019-10-29:
I got this error right out of the box. I tried wiping out my catkin-ws for my project, wiping out my carla-ros-bridge, re-downloading and building carla-ros-bridge, and then re-building my project -- but still no luck. I guess now I'll try the same thing, but downloading your specific version.

javieryu, 2019-10-29:
Issue #203 may be related in some way. 

How you installed/reference the Python API may also be related. Did you use easy_install or are you referencing it directly by sourcing the egg file? Also what version of Python are you using?

chipkajb, 2019-10-29:
I'll take a look at Issue #203 .
I am referencing the Python API by sourcing the egg file. And I am using Python 2.7.

I was able to get it to work without using cv_bridge, but unpacking it directly using struct and then converting it to a numpy array. But this way is much slower (~600 ms). The code I used is below.

`img = np.array(struct.unpack('i'*image.height*image.width, image.data))
img = img.reshape((image.height,image.width))
img = img*255/1000.0
img = img.astype('uint8')`

javieryu, 2019-10-29:
Sourcing the egg file directly seems to be troublesome. I would try it installing it with easy_install/pip in a virtual env.

chipkajb, 2019-10-29:
I see. I will try that.

chipkajb, 2019-10-30:
My issue is resolved. This [page](http://carla.org/2019/07/12/release-0.9.6/) stated that the following fix was made in Carla 0.9.6.

> Fixed std::bad_cast when importing other libraries, like tensorflow, before carla

So that made me think the order in which I import the Python libraries matters. So I tried different orders and the one that ended up working was when I put `import carla` last in my list of imports. Not sure why that matters, but I'm just glad it's working now.

closed 2019-10-30 by javieryu