carla-simulator/ros-bridge #197: rosbridge Launch - Cannot import WalkerControl

aravind93raja, 2019-09-25:
I have set the python path and also sourced the catkin_ws in carla ros bridge,
Then I get the following error when I run the launch file for ROS Bridge

How do i fix this ?  

File "/home/deeplearner/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py", line 40, in <module>
    from carla_ros_bridge.object_sensor import ObjectSensor
  File "/home/deeplearner/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/object_sensor.py", line 14, in <module>
    from carla_ros_bridge.walker import Walker
  File "/home/deeplearner/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/walker.py", line 17, in <module>
    from carla import WalkerControl
ImportError: cannot import name WalkerControl
[carla_ros_bridge-2] process has died [pid 16238, exit code 1, cmd /home/deeplearner/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py __name:=carla_ros_bridge __log:=/home/deeplearner/.ros/log/cdcad5e6-df8a-11e9-9649-d4258b01d015/carla_ros_bridge-2.log].
log file: /home/deeplearner/.ros/log/cdcad5e6-df8a-11e9-9649-d4258b01d015/carla_ros_bridge-2*.log

aravind93raja, 2019-09-26:
[image]

closed 2019-11-13 by fpasch

onebitme, 2020-04-05:
Hello, I am having a similar problem with @aravind93raja 
`    from carla_msgs.msg import CarlaTrafficLightStatus, CarlaTrafficLightInfo
ImportError: cannot import name CarlaTrafficLightInfo
`

fpasch, 2020-04-06:
@onebitme Please call ```git submodule update```.

dmrly, 2020-05-06:
Hello @onebitme    ` from carla_msgs.msg import CarlaTrafficLightStatus, CarlaTrafficLightInfo ImportError: cannot import name CarlaTrafficLightInfo`  I am getting this error too. Could you fix it?

MangalDeep1, 2020-05-09:
@fpasch I am new to Carla and ROS, I am having a similar problem as @onebitme .  I tried git submodule update but did not help. May be I am missing few required steps. Could you please list all the steps to resolve it?

onebitme, 2020-05-11:
@path-env  I've commented out the traffic lights for a couple of weeks...

m-ayala, 2020-05-28:
@onebitme @dmrly @path-env I've had the same issue, looks like it got fixed by git cloning this folder `https://github.com/carla-simulator/ros-carla-msgs.git` 
to /carla-ros-bridge/catkin_ws/src/ros-bridge/carla-msgs

luoxinkaikevin, 2022-04-27:
> @onebitme @dmrly @path-env I've had the same issue, looks like it got fixed by git cloning this folder `https://github.com/carla-simulator/ros-carla-msgs.git` to /carla-ros-bridge/catkin_ws/src/ros-bridge/carla-msgs

I try to  use this method but when I run "roslaunch carla_ros_bridge carla_ros_bridge.launch", the problem still exists. can you help me. thanks for your attention.