carla-simulator/ros-bridge #198: The vehicle model

kargarisaac, 2019-09-26:
Hi,
I'm reviewing Carla and LGSVL o select one of them as a default simulator to be used with Autoware. I cannot find the car model for Autoware. Is it important to use the same car as Autoware? What is the car model that carla-autoware uses? I saw Lexus and jaguar models in LGSVL. Are there such models in Carla? 
Thank you

fpasch, 2019-10-02:
You can select a vehicle via ros-launch https://github.com/carla-simulator/ros-bridge/blob/master/carla_ros_bridge/launch/carla_ros_bridge_with_example_ego_vehicle.launch#L6
By default a random vehicle is selected.

The available vehicles are listed [here](https://carla.readthedocs.io/en/latest/bp_library/)

closed 2019-10-02 by fpasch

anavc97, 2021-01-04:
Hello, I'm trying to define a default model, so as mentioned by @fpasch I changed the "vehicle_filter" default to "vehicle.tesla.model3". It didn't change anything - when I run the carla_ros_bridge_with_example_ego_vehicle package the vehicle still spawns a random model.

I'm using:
Ubuntu 20
Ros Noetic
Python 3.7/3.8
Carla 0.9.9
Carla Ros Bridge 0.9.10.1

I understand that this version of ros bridge is supposed to be used with carla 0.9.10 but I've tried the 2 together and there wer some bugs. This version with Carla 0.9.9 has been working fine. Might this be an incompatibility issue?
Thanks

fpasch, 2021-01-05:
@anavc97 Please use carla 0.9.10.1 as this is the only supported combination and either use `vehicle_filter:='vehicle.tesla.model3'` in the commandline or adapt it [here](https://github.com/carla-simulator/ros-bridge/blob/master/carla_ros_bridge/launch/carla_ros_bridge_with_example_ego_vehicle.launch#L10)