carla-simulator/ros-bridge #203: carla_ros_bridge fails to launch - returns "std:bad_cast"

exoticDFT, 2019-10-24:
I'm trying to launch carla_ros_bridge, but run into an error the instant I attempt to run any of the launch files (carla_ros_bridge.launch, ros_bridge_client.test, etc.) As soon as the ros_bridge launches, it crashes with a "std:bad_cast" error.

All Carla software I'm using are version 0.9.5. I've grabbed and extracted the zips for carla (tag: 0.9.5) and ros_bridge (tag: 0.9.5).

Steps I've taken:

- Compile ros_bridge using catkin_make
- Source the setup.bash file (independently, from devel and my final installation directories)
- Run the CarlaUE4.sh script
- Export PYTHONPATH to include the python2 egg file in PythonAPI/carla/dist
- Run `rostest carla_ros_bridge ros_bridge_client.test` OR `roslaunch carla_ros_bridge carla_ros_bridge.launch`

Regardless of the script called, I always get an error related to "std:bad_cast". I'm attaching the terminal output from the runs, as well as the log for the `rostest` run. Since the error seems to throw as soon as the client attempts to connect, I'm guessing there is some problem with an API version mismatch, but the logs and error messages seem to give no real clues to the root cause.

Any help would be appreciated. Thanks!

[carla_ros_bridge-bad_cast.txt](https://github.com/carla-simulator/ros-bridge/files/3769780/carla_ros_bridge-bad_cast.txt)
[rostest-DNab40a3f8.stanford.edu-9355.log](https://github.com/carla-simulator/ros-bridge/files/3769784/rostest-DNab40a3f8.stanford.edu-9355.log)

exoticDFT, 2019-10-25:
I started to run through the client.py script via pdb and was unable to get the debugger to hit a breakpoint even at the beginning of main. So I stepped through from the beginning and found that the bad_cast is happening when the `CarlaRosBridge` is imported.

exoticDFT, 2019-10-31:
After some more debugging, I have found two potential solutions to get past the issue. However, neither is really solving the issue.

1. If you install the Carla Python API through pip, i.e. `pip2 install carla==0.9.5`, this will stop forcing the "std:bad_cast". However, in my environment, when I use this with some ros code I have in one repo, I get a new issue of "free(): invalid pointer".

2. Using easy_install, i.e. `easy_install --user /path/to/carla/PythonAPI/carla/dist/carla-0.9.5-py2.7-linux-x86_64.egg` or referencing this .egg file location in the python script, I get the "std:bad_cast". To solve it, I have to edit the "client.py" file in ros-bridge to call `import carla` after the ros-bridge specific imports (i.e. make it the last import call.) This solution is related to issue #194, which was referenced the other day.

I'm not sure what's going on here, but it seems the ros-bridge is somehow breaking some basic carla PythonAPI functionality. Maybe ros-bridge is trying to overwrite some methods of the same name or something along those lines? Anyone from Carla comment on what might be happening?

fpasch, 2019-11-13:
Would you be able to test it with Carla 0.9.6?

exoticDFT, 2019-12-16:
@fpasch Sorry for the delayed response. Yes I should be able to do some testing with Carla 0.9.6. As an FYI, I tried using a more recent version of the master branch, maybe about 2 weeks at this point, and was still experiencing the same issues. But please let me know how I can help with resolving this issue.

fpasch, 2020-09-30:
Should be fixed with 0.9.10

closed 2020-09-30 by fpasch