carla-simulator/ros-bridge #204: Invalid message type:carla_msgs/CarlaVehicleControl

janlotus, 2019-10-30:
Hallo,
according to the instruction, :
1. 
cd ~/CARLA
./CarlaUE4.sh
2.
export PYTHONPATH=$PYTHONPATH:~/CARLA/PythonAPI/carla/dist/carla-0.9.6-py2.7-linux-x86_64.egg
source ~/carla-ros-bridge/catkin_ws/devel/setup.bash
roslaunch carla_ros_bridge carla_ros_bridge.launch
3.
cd ~/CARLA/PythonAPI/examples
python manual_control.py --rolename=ego_vehicle
4. then the topic /carla/ego_vehicle/vehicle_control_cmd existiert, and the type is carla_msgs/CarlaEgoVehicleControl , and then I try:
rostopic pub /carla/ego_vehicle/vehicle_control_cmd carla_msgs/CarlaEgoVehicleControl "{throttle: 1.0, steer: 0.0}" -r 10

and there is the Error: Invalid message type:carla_msgs/CarlaVehicleControl,
 I try to use rosmake, but i did not work, there is the warning: the following args could not be parsed as stacks or packages:['carla_msgs']

does anyone know, how to fix it?
i use carla 096 and rosbridge 096,
in version 095, it has the same problem.

closed 2019-11-04 by fpasch

MichaelKlin, 2020-06-13:
Hi @javieryu , i also call python manual_control.py --rolename=ego_vehicle, bat i get black pygame windows. What can it be?

javieryu, 2020-06-13:
Hi @MichaelKlin this is a closed issue for what sounds like a different problem. Please open a new issue if you haven't found a solution yet.