carla-simulator/ros-bridge #206: question about carla_waypoint_publisher

atinfinity, 2019-11-13:
## System information (version)
- Ubuntu 16.04 64bit
- ROS Kinetic
- CARLA v0.9.6(compiled version)
- ros-bridge https://github.com/carla-simulator/ros-bridge/commit/28d4ef607e07d217b873e35aee768d0a11b1bfa5

## Detailed description
I tried to use carla_waypoint_publisher. 
I think that carla_waypoint_publisher publish unintended waypoint.
Is this intended behavier?

### Case1
First time, I don't set initial pose using RViz. But, carla_waypoint_publisher publish waypoint.

[image]

### Case2
I set initial pose and goal using RViz. But, carla_waypoint_publisher publish unintended waypoint.

[image]

## Steps to reproduce
### Case1
```
$ ./CarlaUE4.sh -windowed -ResX=160 -ResY=120
$ roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch
$ roslaunch carla_waypoint_publisher carla_waypoint_publisher.launch
```

### Case2
```
$ ./CarlaUE4.sh -windowed -ResX=160 -ResY=120
$ roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch
$ roslaunch carla_waypoint_publisher carla_waypoint_publisher.launch
```

And, 

- Set Initial Pose using RViz
- Set Goal using RViz

fpasch, 2019-11-13:
That's intended behavior.
See in [code](https://github.com/carla-simulator/ros-bridge/blob/master/carla_waypoint_publisher/src/carla_waypoint_publisher/carla_waypoint_publisher.py#L54): Initially the first spawn_point is used.

atinfinity, 2019-11-16:
@fpasch Thank you for your reply.

> See in [code](https://github.com/carla-simulator/ros-bridge/blob/master/carla_waypoint_publisher/src/carla_waypoint_publisher/carla_waypoint_publisher.py#L54): Initially the first spawn_point is used.

In this case, start and goal are same position. 
So, many waypoints should not be generated.

Currently carla_waypoint_publisher adds offset to calculate a route from the current location to goal. Is there the reason for this value?
https://github.com/carla-simulator/ros-bridge/blob/28d4ef607e07d217b873e35aee768d0a11b1bfa5/carla_waypoint_publisher/src/carla_waypoint_publisher/carla_waypoint_publisher.py#L77

I made simple validation code.

```python
import carla
from agents.navigation.global_route_planner import GlobalRoutePlanner
from agents.navigation.global_route_planner_dao import GlobalRoutePlannerDAO

host = "127.0.0.1"
port = 2000

carla_client = carla.Client(host=host, port=port)
carla_client.set_timeout(2)
world = carla_client.get_world()

for actor in world.get_actors():
    if actor.attributes.get('role_name') == 'ego_vehicle':
        ego_vehicle = actor
        break

location = ego_vehicle.get_location()
print(location)

map = world.get_map()
dao = GlobalRoutePlannerDAO(map)
grp = GlobalRoutePlanner(dao)
grp.setup()

z_offset = 2.0
start = location
end   = carla.Location(location.x, location.y, location.z + z_offset)
route = grp.trace_route(start, end)
print("number of waypoint = {}".format(len(route)))
```

In this code, GlobalRoutePlanner generated 274 points(Depends on current location and map).
I changed `z_offset` to zero, GlobalRoutePlanner generated 1 points.

As a result, I think that this is a problem is to set goal in carla_waypoint_publisher.

fpasch, 2019-11-18:
That offset was added because there were issues with spawning into the ground. That might be solved with newer CARLA versions.

Although you might be right, this is currently more a "feature" than a bug. Because for demo purposes, the car should start driving by default.

What exactly do you want to achieve?
Maybe it would make sense the add the goal to the [launch file](https://github.com/carla-simulator/ros-bridge/blob/master/carla_waypoint_publisher/launch/carla_waypoint_publisher.launch). (Similar to the ego vehicle spawn point, that can be specified [here](https://github.com/carla-simulator/ros-bridge/blob/master/carla_ego_vehicle/launch/carla_example_ego_vehicle.launch).

closed 2020-03-04 by fpasch