carla-simulator/ros-bridge #207: carla_manual_control dies when publishing a new /carla/ego_vehicle/initialpose

bbrito, 2019-11-19:
I am the topic /carla/ego_vehicle/initialpose to reset the vehicle simulation on Carla. However when I do it the node dies and it outputs the following msg:

Traceback (most recent call last):
  File "/home/bbrito/catkin_ws/src/ros-bridge/carla_manual_control/src/carla_manual_control/carla_manual_control.py", line 598, in <module>
    main()
  File "/home/bbrito/catkin_ws/src/ros-bridge/carla_manual_control/src/carla_manual_control/carla_manual_control.py", line 587, in main
    world.render(display)
  File "/home/bbrito/catkin_ws/src/ros-bridge/carla_manual_control/src/carla_manual_control/carla_manual_control.py", line 136, in render
    self.hud.render(display)
  File "/home/bbrito/catkin_ws/src/ros-bridge/carla_manual_control/src/carla_manual_control/carla_manual_control.py", line 463, in render
    rect = pygame.Rect((bar_h_offset, v_offset + 8), (f * bar_width, 6))
TypeError: Argument must be rect style object

Any idea how to fix this?

Thank you so much.

Best,
Bruno

fpasch, 2019-11-20:
My first guess is that while respawning the ego vehicle, some values might still be invalid.
Could you please check the content of ```f```? As this is only the visualization of some numerical value as a bar, a zero value could be used in such cases.

bbrito, 2019-11-21:
I printed f and its value it's always 0.0.
I fixed this issue by forcing f=0.0 and it works. But I guess this is not the proper solution.

fpasch, 2020-03-04:
resolved by #235

closed 2020-03-04 by fpasch