carla-simulator/ros-bridge #217: Compilation error to compile the ros-bridge. I receive "Invoking "make -j8 -l8" failed" error. Please help.

aizzat, 2020-01-18:
Compilation error to compile the ros-bridge. I receive "Invoking "make -j8 -l8" failed" error. Please help. 

(base) irov@irov-ThinkPad-X395:~/carla-ros-bridge/catkin_ws$ rosdep install --from-paths src --ignore-src -r
#All required rosdeps installed successfully
(base) irov@irov-ThinkPad-X395:~/carla-ros-bridge/catkin_ws$ catkin_make
Base path: /home/irov/carla-ros-bridge/catkin_ws
Source space: /home/irov/carla-ros-bridge/catkin_ws/src
Build space: /home/irov/carla-ros-bridge/catkin_ws/build
Devel space: /home/irov/carla-ros-bridge/catkin_ws/devel
Install space: /home/irov/carla-ros-bridge/catkin_ws/install
Creating symlink "/home/irov/carla-ros-bridge/catkin_ws/src/CMakeLists.txt" pointing to "/opt/ros/melodic/share/catkin/cmake/toplevel.cmake"
####
#### Running command: "cmake /home/irov/carla-ros-bridge/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/irov/carla-ros-bridge/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/irov/carla-ros-bridge/catkin_ws/install -G Unix Makefiles" in "/home/irov/carla-ros-bridge/catkin_ws/build"
####
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/irov/carla-ros-bridge/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/melodic
-- This workspace overlays: /opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/irov/carla-ros-bridge/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.19
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 7 packages in topological order:
-- ~~  - carla_ros_bridge
-- ~~  - carla_msgs
-- ~~  - carla_ackermann_control
-- ~~  - carla_ego_vehicle
-- ~~  - carla_manual_control
-- ~~  - carla_waypoint_publisher
-- ~~  - pcl_recorder
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'carla_ros_bridge'
-- ==> add_subdirectory(ros-bridge/carla_ros_bridge)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'carla_msgs'
-- ==> add_subdirectory(ros-bridge/carla_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
[... 83 lines cut ...]
[  2%] Built target _carla_ackermann_control_generate_messages_check_deps_EgoVehicleControlCurrent
[  4%] Generating Javascript code from carla_msgs/CarlaEgoVehicleControl.msg
Scanning dependencies of target carla_msgs_generate_messages_py
[  5%] Generating Javascript code from carla_msgs/CarlaCollisionEvent.msg
[  7%] Generating Javascript code from carla_msgs/CarlaLaneInvasionEvent.msg
[  8%] Generating Javascript code from carla_msgs/CarlaEgoVehicleInfoWheel.msg
[ 10%] Generating Javascript code from carla_msgs/CarlaEgoVehicleInfo.msg
[ 11%] Generating Javascript code from carla_msgs/CarlaMapInfo.msg
[ 13%] Generating Python from MSG carla_msgs/CarlaEgoVehicleStatus
[ 16%] Building CXX object ros-bridge/pcl_recorder/CMakeFiles/pcl_recorder_node.dir/src/PclRecorder.cpp.o
[ 16%] Generating Python from MSG carla_msgs/CarlaEgoVehicleControl
[ 17%] Generating Python from MSG carla_msgs/CarlaLaneInvasionEvent
[ 20%] Generating Python from MSG carla_msgs/CarlaEgoVehicleInfoWheel
[ 20%] Generating Python from MSG carla_msgs/CarlaCollisionEvent
[ 22%] Building CXX object ros-bridge/pcl_recorder/CMakeFiles/pcl_recorder_node.dir/src/main.cpp.o
[ 22%] Built target carla_msgs_generate_messages_nodejs
[ 23%] Generating Python from MSG carla_msgs/CarlaEgoVehicleInfo
[ 25%] Generating Python from MSG carla_msgs/CarlaMapInfo
[ 26%] Generating Python msg __init__.py for carla_msgs
[ 26%] Built target carla_msgs_generate_messages_py
[ 27%] Linking CXX executable /home/irov/carla-ros-bridge/catkin_ws/devel/lib/pcl_recorder/pcl_recorder_node
[ 27%] Built target pcl_recorder_node
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

fpasch, 2020-01-20:
`ModuleNotFoundError: No module named 'rospkg'`
The issue seems to be related to your system setup. Have you sourced your ros-setup-script?

As this is not related to the ros-bridge, I'll close the issue.

closed 2020-01-20 by fpasch

Kvvvv-yzk, 2024-09-08:
我也是这样，你解决了吗