carla-simulator/ros-bridge #231: carla_ros_bridge dies when publishing a new /carla/ego_vehicle/initialpose

bbrito, 2020-02-06:
This problem is similar to one of my previous issues. If I reset the simulation several times using the /carla/ego_vehicle/initialpose topic, the carla_ros_bridge dies giving the following error:

terminate called after throwing an instance of 'std::bad_weak_ptr'

I tried to debug the node, but the same error happens in different parts of the code. Any idea why? Or a possible solution?

Thank you so much for your help.

fpasch, 2020-02-10:
I saw this error also a few times. Until now, I don't have a solution. I suspect, that the ros bridge does not clean up exactly as the Carla Python Library expects it.

For your use case, it might make sense to rework carla_ego_vehicle.py to not respawn the actor (with all the sensors) but to simply move it with ```actor.set_transform()```.
It would be great if you would share your work afterwards by sending a pull request.

bbrito, 2020-02-10:
Thanks! I will try that!

bbrito, 2020-02-14:
I did a pull requested with the fix for this issue.

Best,
Bruno

closed 2020-02-14 by bbrito