carla-simulator/ros-bridge #236: is this bridge only compatible with ros kinetic?

xgnit, 2020-02-28:
I tried to compile the project, then I noticed this step: 

source /opt/ros/kinetic/setup.bash

Then it requires at the following step the package ackermann_msg, which does not exist in my ros distribution melodic.

It means, I should switch my ros to kinetic?

fpasch, 2020-02-28:
There are already a lot of people using ros-bridge with ROS melodic.
Only the documentation is not yet updated.

Have you executed?

    rosdep update
    rosdep install --from-paths src --ignore-src -r

xgnit, 2020-02-28:
hey, I did it step by step copy & paste. The only thing that not totally identical with your instruction is i have to run 

source /opt/ros/melodic/setup.bash 

instead of 

source /opt/ros/kinetic/setup.bash

And it still shows the ackermann_msgs is missing. (tried it again minutes ago)

xgnit, 2020-02-28:
Ah I just checked the log, it seems something is missing by executing rosdep install --from-paths src --ignore-src -r:

E: Unable to locate package ros-melodic-ainstein-radar-msgs
executing command [sudo -H apt-get install ros-melodic-ackermann-msgs]
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-melodic-ackermann-msgs
executing command [sudo -H apt-get install ros-melodic-derived-object-msgs]
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-melodic-derived-object-msgs
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install ros-melodic-ainstein-radar-msgs] failed
  apt: command [sudo -H apt-get install ros-melodic-ackermann-msgs] failed
  apt: command [sudo -H apt-get install ros-melodic-derived-object-msgs] failed
  apt: Failed to detect successful installation of [ros-melodic-ainstein-radar-msgs]
  apt: Failed to detect successful installation of [ros-melodic-ackermann-msgs]
  apt: Failed to detect successful installation of [ros-melodic-derived-object-msgs]

xgnit, 2020-02-28:
For anyone who is interested, I reset my Ubuntu( re-import the disk image), this time it worked.
Not sure what caused the problem, but I might have done something that messed up my ros.

closed 2020-02-28 by xgnit

MichaelKlin, 2020-06-06:
@xgnit by Bridge installation, is using source /opt/ros/kinetic/setup.bash, but can i use melodic, because i have to test in melodic? Thanks you)