carla-simulator/ros-bridge #237: Error when starting ros-bridge within virtual machine

xgnit, 2020-03-02:
I am running Carla on windows 10 host machine and running the ros-bridge on a Ubuntu virtual machine.
I set the shared folder for the VM, so PYTHONPATH will have the egg file from the shared carla folder of the host machine.
(One thing I notice is the python version under the catkin_ws/devel/lib is 2.7 and the python on my host machine is 3.7, they have to be identical?)
Then I remapped the local host of the launch file, change the 'localhost' to '10.0.2.2'.

When I tried to start the launch file, I got the error that python cannot import Carla. How can I fix this? 
Followings are the error log:

PARAMETERS
 * /carla/ego_vehicle/role_name: ['hero', 'ego_veh...
 * /carla/fixed_delta_seconds: 0.05
 * /carla/host: 10.0.2.2
 * /carla/port: 2000
 * /carla/synchronous_mode: False
 * /carla/synchronous_mode_wait_for_vehicle_control_command: True
 * /rosbag_fname: 
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /use_sim_time: True

NODES
  /
    carla_ros_bridge (carla_ros_bridge/bridge.py)

auto-starting new master
process[master]: started with pid [5382]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 192747b4-5c76-11ea-8683-08002762f37a
process[rosout-1]: started with pid [5400]
started core service [/rosout]
process[carla_ros_bridge-2]: started with pid [5406]
Traceback (most recent call last):
  File "/home/vagrant/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py", line 23, in <module>
    import carla
  File "/mnt/carla/PythonAPI/carla/dist/carla-0.9.5-py3.7-win-amd64.egg/carla/__init__.py", line 8, in <module>
    
  File "/mnt/carla/PythonAPI/carla/dist/carla-0.9.5-py3.7-win-amd64.egg/carla/libcarla.py", line 7, in <module>
  File "/mnt/carla/PythonAPI/carla/dist/carla-0.9.5-py3.7-win-amd64.egg/carla/libcarla.py", line 6, in __bootstrap__
ImportError: /home/vagrant/.cache/Python-Eggs/carla-0.9.5-py3.7-win-amd64.egg-tmp/carla/libcarla.cp37-win_amd64.pyd: invalid ELF header
[carla_ros_bridge-2] process has died [pid 5406, exit code 1, cmd /home/vagrant/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py __name:=carla_ros_bridge __log:=/home/vagrant/.ros/log/192747b4-5c76-11ea-8683-08002762f37a/carla_ros_bridge-2.log].
log file: /home/vagrant/.ros/log/192747b4-5c76-11ea-8683-08002762f37a/carla_ros_bridge-2*.log

xgnit, 2020-03-02:
I manage to solve this issue by replacing the egg file of the host machine with egg file of a Linux VM.

closed 2020-03-02 by xgnit

MichaelKlin, 2020-06-04:
Hi @xgnit, i also install Carla on Windows 10 and want to use ROS-Bridge. I install Ros Melodic on my WSL in Windows. Is your methode working? Can you plese maybe describe, what you have done?

xgnit, 2020-06-04:
@MichaelKlin I do not know how it works with WSL, but with virtualbox it worked well(I have melodic as well). Everything should be done according to the official tutorial, just when you need to start the ros, remember to change the ip of the server in the launch file.

MichaelKlin, 2020-06-04:
@xgnit  ok) you install melodic on your virtual box and install ros bridge according to this tutorial: https://carla.readthedocs.io/en/latest/ros_installation/
And i have to change ip like this -> client = carla.Client('new ip', 2000)
Should i share egg.file or to change something in Virtual Box?
If i forget something, write please)

xgnit, 2020-06-04:
@MichaelKlin 
I used this :
https://github.com/carla-simulator/ros-bridge

no, you have to change the ip in the ros launchfile, e.g. carla_ros_bridge_with_example_ego_vehicle.launch etc.

For the egg file you have to use the linux version(egg file is something like binary file so it's platform dependent).

MichaelKlin, 2020-06-04:
@xgnit Thank you) this egg file for linux should i put in linux, in ros-bridge order?

xgnit, 2020-06-04:
@MichaelKlin I am not sure with WSL, but for virtualbox just leave it in Linux so the ros bridge can find it.
You have to understand every step in the official tutorial, most importantly the egg file is for: 
export PYTHONPATH=$PYTHONPATH:<path-to-carla>/PythonAPI/carla/dist/carla-<carla_version_and_arch>.egg

because eventually the ros bridge will use this egg file via this PYTHONPATH variable.

MichaelKlin, 2020-06-06:
@xgnit sorry, last question your carla running only on windows, not virtual mashune? And your ros on virtual mashine?

xgnit, 2020-06-06:
@MichaelKlin yes, Carla was running on windows

MichaelKlin, 2020-06-06:
@xgnit but not virtual mashine, because you write "I am running Carla on windows 10 host machine", and i confused a liitle bit

MichaelKlin, 2020-06-07:
@xgnit hi, it semms to run, but than i get this mistake:
[INFO] [1591551553.875643, 0.000000]: listening to server 10.0.2.2:2000
[INFO] [1591551553.876579, 0.000000]: using vehicle filter: vehicle.*
[INFO] [1591551553.878127, 0.000000]: Waiting for CARLA world (topic: /carla/world_info)...
[INFO] [1591551554.115166, 0.000000]: Trying to connect to 10.0.2.2:2000
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 4dc4cb81 
WARNING: Simulator API version  = 0.9.9-1-g67684c5b 
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc

Do you know, what can it be?

yuanhuiqi, 2022-05-06:
Hello @EvgKlin Have solved this problem? Actually i am also trying to use the ROSbridge in WSL. But it looks like it cant find the Carla in Windows 10