carla-simulator/ros-bridge #240: How to write a subscriber with rospy? I cannot get the refenrence of the msg type.

xgnit, 2020-03-06:
I am trying to write a ros subscriber in Python with rospy, the code is simple:

#!/usr/bin/env python

import glob
import os
import sys
import rostopic

topic = "/carla/ego_vehicle/camera/semantic_segmentation/front/image_segmentation"
data_type = rostopic.get_topic_type(topic, blocking=False)[0]

def callback(data):
	rospy.loginfo(rospy.get_caller_id() + "I heard {}".format(data.data))
def listener():
	rospy.init_node('ros_listener', anonymous=True)
	rospy.Subscriber(topic, data_type, callback)
	rospy.spin()

if __name__ == '__main__':
	listener()

Then I got the error: data_class sensor_msgs/Image is not a class.
It seems python cannot dynamically load the data type, I tried to import the egg file, but the class carla does not have any object called sensor_msgs.

I noticed some py files unter the carla_ros_bridge imported the sensor_msgs, but I am clueless, where does the package live? How can I the a reference to it?

Thnx in advance.

fpasch, 2020-03-06:
It's a plain ros package:
http://wiki.ros.org/sensor_msgs

Something like that should work:
```
import rospy
from sensor_msgs.msg import Image

rospy.Subscriber("/carla/ego_vehicle/camera/semantic_segmentation/front/image_segmentation, Image, callback)
```

closed 2020-03-06 by fpasch

xgnit, 2020-03-06:
it does work! thank you!