carla-simulator/ros-bridge #241: Stop running after a while

tomato2957, 2020-03-06:
I ran the ros bridge with my ros code, using Carla's synchronous mode. At the beginning, all the code was running correctly, but after a period of time, the ros bridge node stopped running and prompted: _Unable to register with master node [http: // localhost: 11311]: master may not be running yet. Will keep trying._

However, the ros master is running correctly at this time. When I use the _rosnode list_ and rostopic list to print related information in the new terminal, the information is correct. The carla ros bridge can still be seen in the rosnode list. 

When I only restart the bridge node, the entire system, including my own ros node, can continue to run. I printed the relevant information and found that when this bug appeared, it was when the bridge node ran to the _self._update (frame, world_snapshot.timestamp.elapsed_seconds)_ sentence in the __synchronous_mode_update function._

Now I can reproduce this bug every time. This bug will appear in my code every hour or so. After restarting the bridge node, it can run for about another hour. 

This bug does not seem to occur often when I do not spawn or destroy actors. I spawn and destroy the actors through another ros node. I am not sure if the operation of the actor will affect the performance of the bridge? This is a problem that confuses me, because all the code works well in the beginning, but the bridge stops running after a certain period of time.

My carla version is 0.9.6, the bridge version used is 138b2f8813e3a2246fe8483932aa0a2d28aa7563, System version: ubuntu18.04
ROS version: melodic

Thank you very much!

fpasch, 2020-03-09:
Could you please check if you set [synchronous_mode_wait_for_vehicle_control_command](https://github.com/carla-simulator/ros-bridge/blob/master/carla_ros_bridge/config/settings.yaml#L10) to enabled?

tomato2957, 2020-03-09:
> Could you please check if you set [synchronous_mode_wait_for_vehicle_control_command](https://github.com/carla-simulator/ros-bridge/blob/master/carla_ros_bridge/config/settings.yaml#L10) to enabled?

Yes,it seems to be set enabled by default. When I print  
      (self.parameters["synchronous_mode_wait_for_vehicle_control_command"])) in bridge.py, it tells me the parameter is true

fpasch, 2020-03-09:
In that case, the ros bridge only continues, if a vehicle control command is received.
Please either switch this parameter to ```false``` or ensure, that you send exactly one vehicle control command per cycle.
With ```false``` the ros bridge will continue with the next cycle, after all sensor data was received. You need to check yourself, that your nodes are able to cope with that behavior.

closed 2020-03-24 by fpasch