carla-simulator/ros-bridge #259: Error when launching carla ros bridge using source repository

arstek131, 2020-04-14:
Following Carla doc I used option B (using source repositry, because I need to edit those packages).
The problem occurs when I start the ROS bridge, I got the error ah shown in the video.
I also managed to install the bridge using option A (apt repository) and everything works well.
That's also the [video](https://drive.google.com/file/d/1Ww4L7BAj69Ma0OYniftm1AFvyhfAV5K_/view)

[image]

vimalrajliya, 2020-04-15:
Did you do `git submodule update --init` before doing `catkin_make`?
This problem might occur when you don't have the carla_msgs package.

arstek131, 2020-04-15:
> Did you do `git submodule update --init` before doing `catkin_make`?
> This problem might occur when you don't have the carla_msgs package.

Thanks problem solved, that was the issue.

closed 2020-04-15 by arstek131