carla-simulator/ros-bridge #263: Current debian missing updated carla_msgs

tbostic32, 2020-04-22:
Installing the most recent package 
```bash
sudo apt install carla-ros-bridge-<melodic or kinetic>
```
and then running (after setting up env) 
```bash
roslaunch carla_ros_bridge carla_ros_bridge.launch
```
I get an error that it cannot import [CarlaTrafficLightInfo](https://github.com/carla-simulator/ros-carla-msgs/blob/cf488bb9bca9e745e3159b1edc884c282d328012/msg/CarlaTrafficLightInfo.msg)

This issue was fixed when I manually cloned, pulled the latest carla_msgs submodule, and then built it manually. I think the package just needs to be rebuilt with the latest carla_msgs version.

fpasch, 2020-04-27:
Please follow https://github.com/carla-simulator/ros-bridge#for-developers if you want to use ros-bridge master branch. This will pull the latest version of carla_msgs.

closed 2020-04-27 by fpasch

tbostic32, 2020-04-27:
I wasn't planning to develop on the carla ros bridge. I was following the instructions for users (https://github.com/carla-simulator/ros-bridge#for-ros-kinetic-users) and they no longer seemed to work since the package that is downloaded and installed does not have the latest version of carla_msgs (running the launch file gives an error that it cannot import CarlaTrafficLightInfo).

Can you confirm that that s no longer a problem for the user installation path?

huangxueqin123, 2020-06-29:
> Installing the most recent package
> 
> ```shell
> sudo apt install carla-ros-bridge-<melodic or kinetic>
> ```
> 
> and then running (after setting up env)
> 
> ```shell
> roslaunch carla_ros_bridge carla_ros_bridge.launch
> ```
> 
> I get an error that it cannot import [CarlaTrafficLightInfo](https://github.com/carla-simulator/ros-carla-msgs/blob/cf488bb9bca9e745e3159b1edc884c282d328012/msg/CarlaTrafficLightInfo.msg)
> 
> This issue was fixed when I manually cloned, pulled the latest carla_msgs submodule, and then built it manually. I think the package just needs to be rebuilt with the latest carla_msgs version.

I have same problems. How to rebuild it.

konradst, 2020-07-18:
@huangxueqin123 https://carla.readthedocs.io/en/latest/ros_installation/