carla-simulator/ros-bridge #268: Lidar PointCloud flipped upside down

Roosstef, 2020-04-29:
Hi all,

since the newest update on Carla 0.9.9 the Lidar Pointcloud ist flipped upside down using the carla_ros_bridge. 

(See attached image)
[image]

Using the Carla_ad_demo the Ground-Truth marker is flipped upside down and the PC is alligned correctly. (See here)
[image]

Is anyone else aware of this issue?

fpasch, 2020-05-04:
Could you please verify with the latest master?

Roosstef, 2020-05-04:
Hi @fpasch,

The NPC Markers are visible now! And the point cloud see to be flipped correctly when in frame "map". But my robot model is flipped upside down.

[image]

When im in frame "ego_vehicle" the PC and the Markers are flipped.
[image]

Thank you for your help.

Greetings 
Stefan

Roosstef, 2020-05-12:
Are there any plans to fix this issue?

fabianoboril, 2020-05-26:
There was an issue with the TF of the ego vehicle, which we just fixed. Could you check the TFs on your side, whether you see any abnormalities? Thank you.

Roosstef, 2020-05-26:
Hey @fabianoboril, there is an issue with the ego_vehicle TF (TF is also flipped upside down.)
How did you fixed this problem? This is a carla_ros_bridge issue right? 
Greetings
Stefan

fabianoboril, 2020-05-26:
@Roosstef We merge a fix today #299 for the ego_vehicle TF.

Roosstef, 2020-05-26:
Ok my bad I missed that :)
Thank you!

closed 2020-05-26 by Roosstef

vedthakur5, 2020-06-03:
@Roosstef How did you set the lidar data on rviz. Can you explain all the steps in brief. My lidar data is not being visible in the rviz

Roosstef, 2020-06-03:
Just Start a Carla Server and run the carla_ad_demo.launch.

vedthakur5, 2020-06-03:
Is it ok to run carla_ad_demo_with_rviz.launch
and  check my lidar sensor:

{
            "type": "sensor.lidar.ray_cast",
            "id": "lidar1",
            "x": 0.0, "y": 0.0, "z": 2.4, "roll": 0.0, "pitch": 0.0, "yaw": 0.0,
            "range": 50,
            "channels": 32,
            "points_per_second": 320000,
            "upper_fov": 2.0,
            "lower_fov": -26.8,
            "rotation_frequency": 20,
            "sensor_tick": 0.0
        },
and what does pointcloud2 "marker" mean.....I know only pointcloud2 which is one of the options in "Add" in Rviz

Please mention the topic of this pointcloud2 also
Thanks for your reply...

vedthakur5, 2020-06-03:
Thanks bro...I solved it myself
Actually in the official carla-ros-bridge pkg the topic for pointcloud2 is:
/carla/ego_vehicle/lidar/front/point_cloud
but since from the below lidar sensor in sensors.json file in carla_ego_vehicle/ config

{
"type": "sensor.lidar.ray_cast",
"id": "lidar1",
"x": 0.0, "y": 0.0, "z": 2.4, "roll": 0.0, "pitch": 0.0, "yaw": 0.0,
"range": 50,
"channels": 32,
"points_per_second": 320000,
"upper_fov": 2.0,
"lower_fov": -26.8,
"rotation_frequency": 20,
"sensor_tick": 0.0
},

we see that the id is lidar1 so the topic for pointcloud2 should be
/carla/ego_vehicle/lidar/lidar1/point_cloud
-----------------------------------------------------------------------------------------------
It's better that the creators change it in official carla-ros-bridge pkg